ArduPilot/ardupilot #4861: Copter: auto mission doesn't disarm if NAV_LAND is not last mission item

tridge, 2016-09-21:
#### Issue details
If there are mission items after a NAV_LAND then the copter won't disarm and will continue with the next mission item. This prevents having a mission with multiple landing points

Naterater, 2019-01-08:
Is it critical that the aircraft disarms after landing if it is not complete with its mission?

rmackay9, 2019-01-09:
In general **not** disarming allows multiple landing point in a single mission.  The way the missions are handled at the moment, once disarmed the mission command processor stops running so if the vehicle disarmed then it would also stop the mission.  I think this request is so that the vehicle disarms and the user then manually triggers a re-arm (in Guided perhaps) and then switches the vehicle back into Auto and issues a mission-start or nav-takeoff command to resume the mission.

There are a couple of potential solutions to support what we do now and what this issue requests:

1. set MOT_SPIN_ARM = 0 so that the props spin down after it's landed.  The MOT_SPOOL_TIME parameter could be increased to slow the spool up time so that the next takeoff is safe (if the spool up time is too low, it might try to takeoff before all the motors have started spinning)
2. add a parameter somewhere (MIS_?) to indicate that the vehicle should disarm after each land.

IamPete1, 2021-08-15:
Fixed by https://github.com/ArduPilot/ardupilot/pull/13721

closed 2021-08-15 by IamPete1