ArduPilot/ardupilot #4865: "suffix is too long in" of AP_Param should be: "prefix is too long in"

dgrat, 2016-09-21:
#### Issue details
Adding AP_Param parameters which have to long prefixes causes a warning about the suffix. 
This is not correct. I suggest to adapt the warning in AP_Param.cpp

    check_group_info:174: suffix is too long in SCALE_LPF
    Bad var table

#### Version
all

#### Platform
all

#### Airframe type
does not matter

#### Hardware type
not important

#### Logs
..

peterbarker, 2016-09-23:
@dgrat It's the combined length which is the problem, of course.

The code is just assuming that the prefix will be of the shortest sensible length, that the suffix is the new thing being added, and that changing the suffix is going to be much less trouble than changing the prefix!

I think that's reasonable?

WickedShell, 2017-11-29:
Peters explanation seems reasonable, and this is mostly a dev related warning who should not have a hard time resolving it with either version printed out.

closed 2017-11-29 by WickedShell