ArduPilot/ardupilot #4866: Where does sim-vehicle.py define what is being built?

mkschreder, 2016-09-21:
#### Issue details
After upgrading to 3.4 which uses new sim_vehicle.py, I'm no longer able to build my sim because it seems that it no longer builds all necessary libraries. 

How has this process changed and where do I need to add my libraries that need to be included into the sim build? 

#### Version
3.4

#### Platform
[  ] All
[  ] AntennaTracker
[x ] Copter
[  ] Plane
[  ] Rover

mkschreder, 2016-09-21:
Also for some reason a frame define that I added to defines.h is not defined for sitl builds and I'm not able to set it by passing CFLAGS either.. (CFLAGS="-DMY_CUSTOM_FRAME" make sitl)

mkschreder, 2016-09-21:
It seems the problem is isolated to the new sim_vehicle.py script because I can build sitl by just invoking make but when I run sim_vehicle, the build process does not build all necessary code.

peterbarker, 2016-09-21:
I think your probelem stems from our change in build system rather than sim_vehicle.py; you can verify this my trying to build directly with waf (by copying/pasting the commands sim_vehicle.py indicates it is using and/or looking at $ARDUPILOT_HOME/BUILD.md

You can ask sim_vehicle.py to use the old make build system with "--build-system=make"

You can tell waf to build your libraries by editing the "wscript" file in the vehicle directory.

Does that clear your problem?

mkschreder, 2016-09-22:
I have added my libraries but now I can not build with custom frame config. I have added my frame to list of frames in wscript but when I try to build it using ./waf --targets bin/arducopter-quad-ptilt I get errors during build stating that QUAD_PTILT_FRAME define is not defined. It should be though considering that the wscript file adds it to defines on the line: defines=['FRAME_CONFIG=%s' % frame_config]

peterbarker, 2016-09-23:
@mkschreder Glad we solved your issue!

closed 2016-09-23 by peterbarker

mkschreder, 2016-09-23:
Was there a target for flashing the firmware in waf?

mkschreder, 2016-09-23:
I'm not able to use the --upload option for some reason when I specify my target. Where can I define it to be the same as arducopter-quad target?

mkschreder, 2016-09-23:
Hmm it looks like it is building sitl board regardless of whether I specify --board=px4-v2 or something else. Why?