ArduPilot/ardupilot #4875: Plane: 3.7.0 User able to spin motor when disarmed

Autonomost, 2016-09-22:
#### Issue details
When the plane is disarmed and on the ground the user is able to spin up the motor by simply reversing the throttle in mission planner. This has also been confirmed in SITL.
#### Version
3.7.0 

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_
Fixed wing
#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_
Pixhawk
#### Logs
_Please provide a link to any relevant logs that show the issue_
https://www.dropbox.com/sh/yli8inghb2ynx6x/AAAJHA72az1cDROwp4z7M9Uma?dl=0

MRazgunas, 2016-09-22:
I think it is normal behavior. When you reverse throttle channel ArduPlane assumes that idle throttle is MAX channel value. Maybe this could be improved with more warnings in GCS side.

magicrub, 2016-09-22:
Firstly, there's nothing the flight controller can do about this with PWM based ESCs. When configuring motors it is expected you are changing the settings to match your ESC. If you are changing them to something else then I'm not sure what behavior you would expect. With a digital ESC, such as UAVCAN, then we should handle this. I don't have an ESC to check, would be nice if someone could do that same test on that hardware.

closing.

magicrub, 2016-09-22:
Also, to be clear, this effects all platforms on all PWM ESCs on all autopilots (not just ArduPilot)

closed 2016-09-22 by magicrub