ArduPilot/ardupilot #4877: Plane: reject lidar data when it disagrees with baro

magicrub, 2016-09-22:
#### Issue details
When landing at same altitude as takeoff, the baro and lidar should somewhat agree during landing. If not, the lidar should fail sanity checks and be ignored and/or reset.

#### Version
3.7

#### Platform
  [  ] All
  [  ] AntennaTracker
  [  ] Copter
  [x] Plane
  [  ] Rover

#### Airframe type
Fixed Wing

#### Hardware type
all

#### Logs
see issue https://github.com/ArduPilot/ardupilot/issues/4876

peterbarker, 2016-09-22:
The LIDAR can help catch baro drift.  I'm not sure the two disagreeing means you don't trust your LIDAR.

magicrub, 2016-09-22:
There are ways

peterbarker, 2016-09-23:
Tom points out that In the log in question, the discrepancy is much greater than any sensible amount of BARO drift.

Naterater, 2019-01-04:
If baro drift is time-dependent, then I am all for this. However fronts moving in, etc, can cause over 50ft of pressure change in 1 hour, so long flight times may have very significant drift. Generally I don't trust this, and I haven't seen recent issues with bad lidar.