ArduPilot/ardupilot #4880: Arduplane: Ground testing RTL and no throttle response

Autonomost, 2016-09-22:
#### Issue details
When testing RTL and Guided mode on the ground do not get any throttle response. When testing takeoff mode on the ground get the correct throttle response. 

#### Version
3.7.0
#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover

#### Airframe type
Fixed Wing
#### Hardware type
Pixhawk
#### Logs
https://www.dropbox.com/sh/yli8inghb2ynx6x/AAAJHA72az1cDROwp4z7M9Uma?dl=0 
[BlueEagleParams.txt](https://github.com/ArduPilot/ardupilot/files/488335/BlueEagleParams.txt)

I have also attached the parameter file.

WickedShell, 2016-09-22:
This is the desired behavior, as you haven't taken off yet (or at a high enough altitude above ground or moving fast enough) throttle is disabled. (It's enabled of course for takeoff, or a manual mode if the aircraft is armed).  Logic is contained here: https://github.com/ArduPilot/ardupilot/blob/d6890495aacd72ba578eecef28c6eb6e9713e815/ArduPlane/Attitude.cpp#L631

closed 2016-09-22 by WickedShell

Autonomost, 2016-09-22:
Okay. This has worked before on the ground with an IC plane?

WickedShell, 2016-09-22:
> ardupilot/blob/d6890495aacd72ba578eecef28c6eb
> 6e9713e815/ArduPlane/Attitude.cpp#L631

@tridge: Any comments about throttle on the ground with a IC engine? (I've
never experimented with that)