ArduPilot/ardupilot #4886: BRD_CAN_ENABLE 2 is not documented on the parameter

WickedShell, 2016-09-24:
#### Issue details
Still working on using a UAVCAN device for the first time, and the tutorial from the device manufcaturer prompetly requests BRD_CAN_ENABLE set to 2, which we haven't documented on the parameter what this does.

I'll try and submit documentation once I figure out what the differences are, but at the moment I'm not sure enough to touch the documentation on this.

#### Platform
[ x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

pavel-kirienko, 2016-09-24:
The values of `BRD_CAN_ENABLE` are encoded here: https://github.com/PX4/Firmware/blob/24e90e3c9054674f191489d236cb962456045b72/src/modules/uavcan/uavcan_params.c#L41-L57

OXINARF, 2016-09-25:
@WickedShell There's a PR to update documentation, #4554, but like I wrote there it adds a value that we don't use in code now.

@pavel-kirienko We don't use the PX4 param interface so, although those values are equal to ours (except 3 that we don't use), they don't have to be.

proficnc, 2016-09-25:
Can 3 sets addressing mode.

proficnc, 2016-09-25:
* @value 0 Disabled
 * @value 1 Enabled
 * @value 2 Dynamic ID/Update
 * @value 3 Motors/Update

OXINARF, 2016-09-25:
@proficnc Please point where in code value 3 is used.

EShamaev, 2017-06-26:
This is outdated by recent changes. I feel should be closed.

closed 2017-06-26 by EShamaev