ArduPilot/ardupilot #4887: Support for triggering the Parrot Sequoia

Jman841, 2016-09-24:
#### Issue details
Currently there is no way to directly trigger the Parrot Sequoia from ardupilot autopilot systems, I'm not exactly sure if there are hardware changes required or if it can only be done through software currently, however, it would be great if it could be triggered directly through arduplane instead of having to use a separate setup/system to trigger the camera.

#### Version
3.7.0

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[  X ] Plane
[  ] Rover

#### Airframe type
All

#### Hardware type
Pixhawk

Naterater, 2019-01-08:
In general this would replace the VP-systems Cam-Remote. I agree it would be incredibly useful, especially for geotagging, etc. I hope someone with the knowledge will pick this issue up some day.

WickedShell, 2019-04-05:
Sequoia is meant to be internally managed on taking photos. It does also support PTP and a HTTP API as well, but neither of these are interface for ArduPilot to manage internally. You can use a companion computer to manage the camera under these use cases.

closed 2019-04-05 by WickedShell