ArduPilot/ardupilot #4889: handle_mission_write_partial_list is too strict

ptsneves, 2016-09-25:
#### Issue details

Currently `GCSCommon.cpp@handle_mission_write_partial_list` checks if the partial waypoint list is inside the current mission count boundary unnecessarily. This limits the utility of partial waypoint writing modifications because we can never expand the mission count.

A very simple case is appending a waypoint to a waypoint list. As far as I know a full rewrite of the mission plan would be required because the only other way to change the waypoint list is sending a `mavlink_msg_mission_count*` message with the previous mission count plus one. This is exactly what a partial list is for.

Removing the upper bound checks would not result in a problem because `GCS_Common.cpp@GCS_MAVLINK::handle_mission_item:839` already handles the case where the mission seq is bigger than mission.num_commands().

Am I wrong in these assumptions? If theoretically there is no problem I can produce a patch removing the upper bound checks.

WickedShell, 2016-09-25:
@ptsneves I'm in agreement with you, it will require some careful testing, but this has been sitting on my local todo list! (I forgot to open an issue on it). If you want to go for it that would be excellent!

You are correct that the upper bound check can be mostly removed. The limitarion is that the lower bound must be directly after the last item we already have, or within the scope of the current mission, you can't allow a gap between the current mission and the new partial upload, but that should be straightforward to be aware of.

Naterater, 2019-01-08:
@WickedShell, any chance this passed your radar recently?