ArduPilot/ardupilot #4894: Copter: Support AP_GPS arming checks

WickedShell, 2016-09-26:
#### Issue details
The u-blox driver audits that it has fully programmed the GPS settings which we expect and need. The arming checks have been in plane for quite a while and well proven there: https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Arming/AP_Arming.cpp#L346

It would be really nice to fit this in the 3.4 release. I thought this had made it into copter a long time ago, but this provides a much better audit of having the GPS settings that were requested, and ensuring that the performance matches the assumptions that the rest of the system are making

I'm not sure if it's to late to get this in @rmackay9 but it would be really nice if at all possible.

#### Platform
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover

rmackay9, 2016-10-05:
somewhat similar to this: https://github.com/ArduPilot/ardupilot/issues/4273

rmackay9, 2016-10-10:
This is resolved with this PR: https://github.com/ArduPilot/ardupilot/pull/4942

This will go out with AC3.4-rc6 this week.

closed 2016-10-10 by rmackay9