ArduPilot/ardupilot #4895: Copter: bug in AC_AttitudeControl.h

umbraclet16, 2016-09-27:
#### Issue details
Function 'get_att_target_euler_cd()'  in AC_AttitudeControl.h
> Vector3f get_att_target_euler_cd() const { return _attitude_target_euler_angle*degrees(100.0f); }

should be:

> Vector3f get_att_target_euler_cd() const { return degrees(_attitude_target_euler_angle)*100.0f; }

#### Version
3.4-dev

#### Platform
 Copter

umbraclet16, 2016-09-27:
I wonder why DesRoll/DesPitch is correct when checking the log file in MP...

rmackay9, 2016-09-27:
I don't think it's technically a bug. I think you'll find the result is the same using both methods.  converting from radians to degrees is just a multiply by (about) 57.  The ordering of multiplication doesn't affect the final answer unless we're hitting an overflow which I don't think we are in this case.
Still, I actually prefer the method suggested above so if someone wants to make a PR to change it, I'll pull it in.

umbraclet16, 2016-09-27:
Yes I shouldn't have called it a bug. I just felt very weird at first sight of the function and didn't think it over... And I wrote this issue rather than a PR because I haven't pulled for months and I don't want to update right now. Never mind then.

magicrub, 2016-09-27:
I can do it..

magicrub, 2016-09-27:
So, this gets complicated. 
_attitude_target_euler_angle is Vector3f
1000.f is float
degrees() will only convert a float to a float

in theory degrees() should accept a vector3f (3 multiplcations) and then we multiple y 100 (another 3 multiplications) but if we do it the origional way it's the same result with only 4 multiplications.

I agree the way it is currently written is a bug but it happens to give us the correct result. I'm open to suggestions on how to improve this without having to do additional math for the same result.

rmackay9, 2016-09-27:
I have to reply because apparently github is blocked in my current location.  

It's really not "bug" because it produces the correct answer and my guess is that it was actually done this way intentionally because it makes it easier to go through later and change all the code to use radians.
There's actually a PR from Sid to change everything to radians.  Maybe we should spend our time on that PR instead and close this issue?

OXINARF, 2016-09-27:
It's not a bug and, as you said, it is a better way because it allows less multiplications. If there isn't a comment in the function let's add one on why it was done that way and explaining that it isn't a bug.

magicrub, 2016-09-28:
Added a comment: https://github.com/ArduPilot/ardupilot/commit/3adf8c3e51371e04ecc0f5a8bc416c5d343e298b

closing...

closed 2016-09-28 by magicrub