ArduPilot/ardupilot #4898: Bug in NavEKF median filtering RangeFinder data

johnhump, 2016-09-28:
#### Issue details
Starting at line 5137, of ver 3.4, master branch, AP_NavEKF_core.cpp, this logic block is incorrect:
if (storedRngMeas[0] > storedRngMeas[1]) {
                minIndex = 1;
                maxIndex = 0;
            } else {
                maxIndex = 0;
                minIndex = 1;
}
Regardless of the if statement, maxIndex is always set to 0 and minIndex is always set to 1.  This is a small and easy change and I'm not motivated to learn the pull request process.

#### Version
First saw on 3.3.2, apparently still exists on 3.4

#### Platform
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

magicrub, 2016-09-28:
Thanks for the find, we'll look into it!

priseborough, 2016-10-19:
Thats in the old EKF - I will check to see if it appears in EKF2

priseborough, 2016-10-19:
Thanks for that @johnhump, it affected the new filter as well. I've submitted a PR.

https://github.com/ArduPilot/ardupilot/pull/5020

rmackay9, 2016-10-19:
Fixed in master and will go out with -rc7 within a week.

closed 2016-10-19 by rmackay9