ArduPilot/ardupilot #4901: Please tag 3.4 RC1, RC2, RC3, RC4, RC5 releases

rrr6399, 2016-09-28:
#### Issue details
Please create tags or branches of the RC* releases for 3.4. Right now, it is difficult to find the correct snapshot to run comparisons against.

#### Version
3.4

#### Platform
[  ] All
[  ] AntennaTracker
[x ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
N/A

#### Hardware type
N/A

#### Logs
N/A

rmackay9, 2016-09-29:
AC3.4-rc1:  dfa90d29bae3870ec0f8353a7a71a2b7b28ea173
AC3.4-rc2:  f46a31d06924035989b0dea842f4897e15d61158
AC3.4-rc3: e2a9772ce158edc2f7b1b36aab34d74dd3209286
AC3.4-rc4: 652840ca06784a1794be050be21d06d3408f8dd0
AC3.4-rc5: 0d9d101fef8bd061497c9892df9fcfe136c3c3fc

Hopefully this gives you want you're looking for.  I'm not sure if we should tag all release candidates... just leads to tons of tags... If a few more people want this we can do it, I know the other vehicles don't do this though.

closed 2016-09-29 by rmackay9

rrr6399, 2016-09-29:
Thanks for that! Its important for us since we're staying with RC1 right now until RC5 is fixed. Its good to know what changed between releases to help debug issues as well.

rrr6399, 2016-09-29:
@rmackay9  Did you push your changes to master?

rrr6399, 2016-09-29:
@rmackay9 Would you add actual tags on github, it looks like there are problems checking out cloned code based on a commit:
https://github.com/travis-ci/travis-ci/issues/617

rrr6399, 2016-09-29:
I was able to work around the fact that the commits weren't coming down to my local clone by forking ardupilot on github, creating the Arducopter-3.4RC1 branch and then cloning that repository:
https://github.com/rrr6399/ardupilot

magicrub, 2016-09-29:
Cloning essentially just gets you a current copy. Forking will allow you to
stay on track with master, as you have found.

rrr6399, 2016-10-04:
Any release (RC or production) of the code should have a tag associated with it. That would eliminate some of the hassles of replicating the snapshot for a given RC release when trying to determine what has changed between releases in order to debug problems discovered with new releases.

magicrub, 2016-10-04:
Totally agree