ArduPilot/ardupilot #4902: UAVCAN GPS will never be found if not connected during vehicle boot

WickedShell, 2016-09-28:
#### Issue details
Steps to reproduce:
BRD_CAN_ENABLE 2
GPS_TYPE2 9 (works on GPS1 as well as 2, I'm just testing with GPS2 in this scenario)
Boot the pixhawk without the can device connected, after the device has completely booted (and reported that the GPS was detected as PX4 (this is a false message, it just means we selected the PX4 driver)) connect the CAN GPS. (I'm testing with a Zubax GNSS2) the GPS will never be detected/found.

If you start with the GPS connected, detect it once then disconnect/reconnect it it will be found correctly.

I'm not yet sure if this is an issue because we are leaning on the PX4 GPS drivers to do the CAN decoding, or something with the UAVCAN driver. This doesn't look like an issue from the GPS library however.

#### Version
Plane 3.7.0

#### Platform
[ x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

#### Hardware type
Pixhawk, Zubax GNSS2

WickedShell, 2016-09-29:
Investigation via NSH reveals that the PX4Firmware layer does find the UAVCAN device as a GNSS correctly when it is connected after the boot process.

magicrub, 2016-09-29:
Any difference in behavior if the uavcan one is the only one in the system?
I mean, system boots with zero GPSs then you plug in the uavcan.

WickedShell, 2016-09-29:
This was with only the UAVCAN GPS connected. (The first GPS wasn't connected). I'll try and reproduce it with just GPS1).

WickedShell, 2016-09-29:
@magicrub you are correct, can't reproduce the problem when it's connected to GPS1, GPS2 is a different story.

pavel-kirienko, 2016-10-31:
I'm not familiar with APM architecture, but what I gathered about half a year ago from conversations with @hsteinhaus is that unlike PX4, APM fundamentally does not support hot-plug sensors. In order to account for that, the module that initializes the PX4 layer holds the boot process until the UAVCAN driver from PX4 has reported that all UAVCAN nodes have been discovered (so that no hot plugging occurs later). The wait is implemented here: https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_BoardConfig/px4_drivers.cpp#L241-L248

Invoked (indirectly) from here: https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/system.cpp#L128-L130

The call goes through https://github.com/PX4/Firmware/blob/master/src/modules/uavcan/uavcan_main.cpp#L1140-L1156 and lands in the libuavcan library at https://github.com/UAVCAN/libuavcan/blob/master/libuavcan/include/uavcan/protocol/dynamic_node_id_server/abstract_server.hpp#L65-L103

hsteinhaus, 2016-11-02:
If I remember correctly, the reason for waiting for the enumeration to be completed was rather the baro sensor than the GPS itself (and may be also the mag - not sure about this point). The baro code tries to open the appropriate device files in it's constructor (https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Baro/AP_Baro_PX4.cpp#L23-L34) and proceeds if they are not instantly available. This led to nasty race conditions - sometimes APM won and sometimes the enumeration did. Hence we implemented the synchronization to get a clean and reliable behavior.

However, I do not see a good reason, why an UAVCAN GPS connected later should not be recognized by the GPS driver. All the driver does is to subscribe via the PX4 ORB - as soon as it receives updates it will forward them to APM. If that does not happen, I can only agree to @WickedShell that the current behaviour is a bug. Unfortunately I am travelling right now and have no hardware here to play a bit - but I will look into this issue as soon as I am home again.

hsteinhaus, 2016-11-02:
Compass is also not ready for hot-plugging: 

https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Compass/AP_Compass_PX4.cpp#L66-L79

WickedShell, 2016-11-07:
@lucasdemarchi have you confirmed that this will is actually resolved for the second GPS instance? Thats the specific error that I was seeing, (and I don't see how the referenced PR could fix that)

lucasdemarchi, 2016-11-07:
I think github automatically closed this issue when I merged the other PR. I didn't even read this issue...

EShamaev, 2017-06-26:
This is outdated by recent changes. I feel should be closed.

PR #5155 (2016-11-07): AP_GPS_PX4: unsubscribe on driver destruction. 
An unlimited number of driver objects is constructed by the AP_GPS class, if the configured driver does not receive data. In a hot-plug scenario, this behavior is exhausting the ORB's resources if the driver does not unsubscribe correctly. This PR adds a destructur that ensures un-subscription. 

Fixes #4902
closed 2017-06-26 by EShamaev