ArduPilot/ardupilot #4909: AP_Notify should be statusing filter status not GPS status

John-Ingersoll, 2016-09-30:
#### Issue details
AP_Notify subclass RGBLED reports the status of GPS to the pilot.  It is assumed that this is directly reflective of the EKF filter performance to allow different nav modes (Alt Hold, Loiter, Auto).

The LED doesn't provide meaningful information to the pilot when flying non-gps configurations such as OF modes.

The LED should status the three capabilities of navigation:
Direct flight capable (stabilize, acro, land no gps  possible)
Altitude control (land with gps, alt hold possible)
full control possible (loiter, auto, RTL, guided possible)

John-Ingersoll, 2016-10-03:
Oops,  mostly duplicative of #2168

magicrub, 2016-10-03:
Closing in favor of https://github.com/ArduPilot/ardupilot/issues/2168

closed 2016-10-03 by magicrub