ArduPilot/ardupilot #4911: Battery: utilize battery ESR in voltage readings

magicrub, 2016-09-30:
Post from Charles Linquist:

the battery mAH counter in APM gets reset every power cycle. I watch the percentage of battery left much more than the battery voltage, because with percentage, I don't have to do any mental arithmetic. I have mistakenly connected a battery that wasn't fully charged, and started a mission. I looked down and the remaining battery percentage was 80%, but the battery voltage was at the discharge level.

I think that the "remaining" percentage should have a voltage component to prevent this.

I have had much better results with my battery voltage readings since I
modified AP BATT_MONITOR.CPP as follows:

     if(_monitoring == AP_BATT_MONITOR_VOLTAGE_ONLY || _monitoring ==
AP_BATT_MONITOR_VOLTAGE_AND_CURRENT) {
         // this copes with changing the pin at runtime
         _volt_pin_analog_source->set_pin(_volt_pin);
         _voltage = (_volt_pin_analog_source->voltage_average() *
_volt_multiplier)+ (_current_amps * _batt_esr);
         if(_volt2_pin_analog_source != NULL) {
             _voltage2 = _volt2_pin_analog_source->voltage_average() *
_volt2_multiplier + (_current_amps * _batt2_esr);     ;
         }
     }

The addition of the _battx_esr   (which I presently have set to .016f)
term  compensates nicely for the ESR voltage drop of my battery
under load. I measured my battery ESR + drop in the wires leading to the
voltage sensor, and found that the resistance is about 16 milliohms.

With that addition, I find that my displayed battery voltage is close to
constant, regardless of the current draw.  During flights, I get a nice
downward slope of
battery voltage regardless of whether I'm climbing or cruising or -
whatever.  Of course, the slope of the line is dependent on the draw,
but it no longer
has big "steps".

With my change, I'm measuring the actual internal CELL voltage, as
opposed to TERMINAL voltage.  My only issue now is that my ESR value is
fixed in the firmware.
Yes, I have variable names, but the values are equated to static values.
I don't know how to add a new parameter to Mission Planner and Mavlink
in order make a variable that I can change as my battery ages.  If you
could help me there,
I would appreciate it.

I was thinking of making this process automatic by measuring the
(uncompensated) voltage and current at a low-current point(like a hover)
and again during a climb.
If the two measurements were taken only a few seconds apart, the ESR
could be calculated by   (Voltage(hover) - Voltage(climb)) /
(Current(climb) - Current(hover))

I should also mention that I added a more effective averager for the
_current_amps variable.  Now my current readings are a bit smoother than
before.  This also
helps with my modified battery voltage routine.

R-Lefebvre, 2016-10-01:
Excellent idea.  I think this could be implemented at a parameter that just defaults to zero.  Users who want to take advantage can set it appropriately.  I wouldn't bother attempting to auto-tune this value.

But, this is simple, effective, and comes from sound engineering theory.

I've seen a lot of attempts to calculate % remaining from the voltage, which is hopeless.  But this at least cleans up our voltage reading.

magicrub, 2016-10-01:
In the above example, the esr would need to be 1.0. Using 0 to disable makes more sense.

lvale, 2016-10-02:
An easy solution to a big problem. I would only add the automatic ESR calculation to be continuous. It could start with a default value, but would adjust during flight, because every battery is different. Setting the default value to zero would disable the calculation, and the ESR adjustment.

magicrub, 2016-10-02:
I agree. I'd like this to be learning and therefore always accurate.

olliw42, 2016-10-03:
guys,

I think you want to be careful and first critically evaluate if the achievements compensate for the drawbacks. As much as I understand it the primary motivation is to have a (roughly) correct battery status, going somewhat in the direction of these smart batteries. 

Now, what was called a "sound engineering theory" isn't anything near that, but just the cheapo poor mans theory. The internal resistance of a lipo is anything than a well defined quantity, which varies depending on the time profile of the load, varies during the discharge, and - most relevant - with age, as well as with the "type" of the battery (e.g. C rate, mAH, brand) (some years ago I did some hobby-grade "research", http://www.olliw.eu/2012/lipo-ri-quality/, http://www.olliw.eu/2011/450er-lipoblog/, but googling for serious work will show you quickly that I'm not talking total nonsense). I just want to mention one implication of that:

Just consider the situation that one is doing two flights in a row using two packs. These two packs might have different age, or they might have different C rating, or different mAH. This would immediately confuse the above approach, since the flight controller has no chance to know the approrpiate resistance to calculate the starting condition of the pack, unless the user has set the correct resistance beforehand, which I seriously doubt anyone finds convenient. Note that also learning doesn't help, since the learned value will be that of the previous pack.

There is a simple reason why smart batteries each have their "own" status evaluation electronics ...

So, the approach might be of use when essentially only identical packs - same brand, same charge, same C rate, same age - are used. If that is not given, I think it has a chance to also seriously confuse users. You will easily identify more "edge" cases. 

I'm not saying you shouldn't do it, but wanted to point out that it is - IMHO - not nearly the final solution to a big problem as which it seems to be perceived. :)

Cheers, Olli

R-Lefebvre, 2016-10-03:
Olli, thanks for pointing out the obvious that all batteries have different ESR, and that it changes with age of the battery.  

All the reasons you mention, is just one subset of the reasons why attempting to guestimate battery capacity based on voltage is hopeless, which has been asked for many times before.

However, having the user enter an ESR is simple, it IS sound engineering theory, and if a conservative value is chosen, the result will help to stabilize the voltage reading at least somewhat.  It cannot be perfect, but it's better than 'zero' which is what we use now.

lvale, 2016-10-03:
Thanks @olliw42 
without going to a "full" smart battery this approach provides a simple way to get better results than what we have now. My suggestion of having a continuous/interval measurement and subsequent calculation would be a way to evaluate any battery, independent of it state and characteristics, as long a "standard" lipo is used. So, giving an initial "reasonable" default value for ESR (per cell) and adjusting along the flight according to the measurements is better than what we have now (ie nothing) :)
I had read your articles a few years ago, and even managed to do some tests when my Orbit Microlader was a good charger :)....

magicrub, 2016-10-03:
How about both options? Set to 0 for disable, -1 for always-on-learning, and any other positive value for a fixed ESR.

magicrub, 2016-10-03:
Now the question is how do we learn it?

lvale, 2016-10-03:
the measurements would be performed initially at a low power consumption point (after arming and if motors spinning) and again on takeoff or when Thr_mid is reached. 
Then repeat measurements(V and A) when at a thr_out pwm_out less than (thr_value+1)+25% pwm_out

magicrub, 2016-10-03:
The other thing to consider is if the computed ESR should persist in eeprom, and if so then we'd want to have another param.

lvale, 2016-10-03:
Absolutely not. Like @olliw42 explained above there are no 2 batteries that behave the same and even the same battery will behave differently on different ocasions. Let's KISS (Keep It Super Simple)

R-Lefebvre, 2016-10-04:
Perfect is the enemy of good.

magicrub, 2016-10-04:
And second place is first loser?

R-Lefebvre, 2016-10-04:
My point is, we could have something simple, that roughly works, easily.

Or, somebody could put a lot of work into developing a complicated algorithm, test it to make sure it works against a range of setups and conditions, support users when it doesn't work.

Who is taking that on? Is Luis or Olli volunteering?  Or will we end up doing nothing, because we nobody takes on this work and we pass over the simple solution?

olliw42, 2016-10-05:
I'm happy to volunteer with implementing the case "0 for disable".

many thx, Rob, for the lesson on what an engineer accepts as a "sound engineering theory". Truth is I'm not an engineer, and am not going to go with that acceptance level, but it's good to know.

no, Luis, no idea, IMHO only case 0 is a good idea :). I would find a prearm check which throws a warning in case the voltage is too low for the battery to be fully charged a good idea though (this can be easily automatized for different cells, needs to know "only" the type of battery, lipo,A123,...).

WickedShell, 2016-10-05:
There is a arming check for minimum voltage, however it is part of the
arming library, so it's not yet a part of copter. You provide the cutoff
voltage which I find much more useful then an automated check (I routinely
fly batteries below 3.7V but I have a very good calibration curve for doing
so)

IamPete1, 2021-07-25:
No update since @amilcarlucas suggests closing, so I'm going to do that...

closed 2021-07-25 by IamPete1