ArduPilot/ardupilot #4912: ArduPlane v3.7.0 - Sends mavlink messages with two different system ids

DonLakeFlyer, 2016-09-30:
#### Issue details
Steps:
* Flash ArduPlane 3.7.0 to Pixhawk
* Boot QGC (with Pixhawk disconnected)
* Connect Pixhawk to QGC via USB

Result:
ArduPlane sends mavlink messages with sysid of 7 and 1. 1 is the correct system id. This causes QGC to think there are two different vehicles connected and shows all it's multi-vehicle ui. Pretty bad bug for any groundstation that supports multiple vehicles. The mavlink messages with id 7 are sent first during board boot I believe, then it switches to 1.

Here is QGC mavlink inspector showing the messages:
[image]

DonLakeFlyer, 2016-09-30:
ArduCopter 3.3 doesn't do this. ArduPlane didn't used to do this in older versions.

mboland, 2016-09-30:
Arducopter does this in 3.4, QGroundControl and Mission Planner

tridge, 2016-10-01:
thanks for letting me know, I'll fix this now

rmackay9, 2016-10-06:
I tried to set this to Copter-3.4 milestone but github doesn't accept two milestones so resetting to plane only.

KIrill-ka, 2016-10-15:
As a workaround, I set MAV sysid to 7 :)

DonLakeFlyer, 2016-10-15:
Ah, thanks for the suggestion! Helpful for my testing.

OXINARF, 2016-10-18:
Fixed in master.

closed 2016-10-18 by OXINARF

rmackay9, 2016-10-18:
Oh, excellent that it's fixed!