ArduPilot/ardupilot #4917: Separate CAM messages: improvement

Naish88, 2016-10-03:
#### Issue details
After a long mapping missions the time required to download the log via mavlink or usb is enormous due to the size of the .log file. Solution will be to create a separate ( a copy ) of all the CAM messages on a different file. 
Example log download:
03092016-0933.log 
03092016-0933-CAM.log

All this is needed because: 

- Multiple missions are flown on a single day, that can lead to confusion of the logs vs pictures if they don't get organised straight away 
- SDcard is not easily accessible all the times, and also the slots are not designed to withstand high numbers of insertions/removals. 
- User should not forced to reduce the data logged to reduce the amount of time needed for download. Logs are critical to troubleshoot issues so more data you have better it is. 
-Planes are now capable of flying for several hours on a mapping mission, making really impossible to even think about downloading the logs over USB

#### Platform
[  ] All
[  ] AntennaTracker
[ x] Copter
[ x] Plane
[ x] Rover

magicrub, 2016-10-03:
One option is to use a Linux based Ardupilot flight controller. The logs
can easily be copied off via Wi-Fi.

WickedShell, 2016-10-03:
Closing in favor of selective downloading of logs. #3187 This will cover the problem you've encountered for this.

Ironically #3817 is a duplicate of #1978 which is also still open... Oops!

closed 2016-10-03 by WickedShell