ArduPilot/ardupilot #4918: Copter: LED should only go green when EKF position is available

rmackay9, 2016-10-03:
AC3.3 & AC3.4 will show a green LED (on Pixhawk) in manual modes once the GPS has a 3D Lock.
In modes that require GPS (Loiter, Auto) the led may be blue if the EKF hasn't yet got a good position estimate (this takes longer because it requires a low velocity reported from the GPS).

This can lead to the situation in which the LED is green while in stabilize but switches to blue if the user switches to Loiter (while disarmed because otherwise the vehicle would refuse to go into Loiter).

During AC3.4 beta testing it was suggested that the LED colour should only go green after the EKF has a good position estimate regardless of whether the mode requires GPS or not.

This will likely lead to support calls, why doesn't the GPS work as well in AC3.4 as it did in AC3.3 but it still might be best.

http://discuss.ardupilot.org/t/rgb-leds-not-showing-true-state-of-gps-mode/11805/8

R-Lefebvre, 2016-10-03:
Why make this an issue?  Are you asking a question or looking for input?

Personally, I think this is a good idea.  We are making an autonomous controller, not a racing/acro controller.

rmackay9, 2016-11-19:
Closing because there's other items covering the same topic.

closed 2016-11-19 by rmackay9