ArduPilot/ardupilot #4919: Copter: allow arming in AUTO

rmackay9, 2016-10-03:
We do not allow arming in AUTO for safety reasons.  In particular, a user may be confused about which flight mode they are in and if they armed in AUTO and raised the throttle the vehicle could immediately fly up before the pilot has time to realise what has happened.

We should allow arming in AUTO if we can be sure the pilot understands what they're doing.  This could be accomplished by accepting an ARM command from the GCS if some field also contains a magic number.  The idea being that the GCS has somehow confirmed with the pilot that they understand they're arming in AUTO.

khancyr, 2016-10-03:
What about sending arm message, and then arm message with magic value before a 5s timeout. That way it ensure that the user realy kown what it does, that is not and error, and that the user doesn't always use magic number as default !

R-Lefebvre, 2016-10-03:
Yep, this is a good idea.  I was wondering just yesterday what the reasoning is for not arming in Auto.  I think the current status is "lesser of evils" because as you say, this is a serious safety problem.

But arming from GCS would be nice.

umbraclet16, 2016-10-03:
I recommend the following:

- Allow arm on GCS unconditionally. Because if the pilot wants to fly in manual mode like loiter, he is not likely to arm with GCS;
- Change RC arm action for AUTO mode. For example, we can arm in AUTO with "throttle down, yaw right, and pitch down". If you want to arm in other modes, just do "throttle down and yaw right". This can prevent arming unintentionally in AUTO mode.

R-Lefebvre, 2016-10-03:
I like that second option.  Allow arming in auto with the special RC Tx gesture.  Throttle down-right, and pitch/roll stick down-left?  Or has that one already been used?

lvale, 2016-10-03:
Check with @jschall because he was looking for a stick combo to start the onboard mag calibration.

ThorstenDP, 2016-10-03:
Instead of introducing too many and maybe confusing TX gestures what about

- an Auto mode arming delay 
- which is accompanied with warning sounds from the buzzer. 

This should give time to disarm or to switch the mode if falsely armed in Auto. 

(As mentioned elsewhere for security reasons I would really appreciate if any potenially dangerous situation like failsafes, ADSB responses, Arming in Auto mode etc. would be accompanied with warning sounds.)

jschall, 2016-10-04:
My recommendation is:
- Allow arming in AUTO mode, if and only if the next mission item is a TAKEOFF.
- Change the "begin mission" trigger to one of the radio switches.
- Provide a parameter option to allow users to begin auto flights by raising the throttle all the way to maximum

Don't add an inordinately long delay. That could create a dangerous situation for users who do not have a buzzer.

ThorstenDP, 2016-10-04:
Yes, that's right. If you don't have a buzzer a delay is maybe not the best idea. My aim was just to keep it simple. No tricky gestures or any annoying additional interaction when you are sure what you are doing and using it frequently.
May be the simplest way would be a "AutoArming" parameter which by default is set to off.

R-Lefebvre, 2016-10-04:
Actually, I like Jonathan's suggestion best of all.

If auto mission starts with a Takeoff command with an Alt at least 5m, then simply allow stick arming with no complicated gesture.  Arming into Auto, and hold at Mot Spin Armed.  Copter will hold in this condition until throttle is raised to max. Then trigger the take-off.  This should make it safer, as presumably if the user is unaware they are in Auto mode, maybe they think they're in Stab or Loiter, they would not stand near the copter and raise the throttle to max.

Then in parallel, allow arming from GCS with the magic number. This would allow an auto flight without needing an RCTx.

jschall, 2016-10-04:
Alternately change the way takeoff works so that when it is manually triggered, the throttle stick functions as in LOITER. That way the user is in total control and can descend. Then, offer a switch and a mavlink message to trigger mission begin and do it all autonomously.

rrr6399, 2016-10-04:
This is what we do for GCS controlled flights:
1.) Arm in Loiter or Stabilize
2.) Switch to guided mode and issue the CMD_NAV_TAKEOFF to 2 m or so
This action provides a sanity check that all systems are working fine with a low risk altitude
3.) Switch to auto mode to execute mission
4.) Switch to Land if needed

Warning: GCS initiated R/C overrides are required for takeoff and landing if the R/C is off.  The firmware defaults the yaw, pitch and roll values to 0 rather than trim (1500) values. Your copter will spin wildly on takeoff or go to max left roll, down pitch and left yaw on landing.)

peterbarker, 2016-10-06:
Note that Randy pointed out that the current implementation of "arm force" bypasses the check for whether the mode is armable :-)  (https://github.com/ArduPilot/ardupilot/pull/4891)

lthall, 2018-01-12:
[Duplicate](https://github.com/ArduPilot/ardupilot/issues/3232)

rmackay9, 2021-03-02:
This is done in master now using the AUTO_OPTIONS parameter so closing.

closed 2021-03-02 by rmackay9

jankrystianserafim, 2023-07-07:
@rmackay9 
Hi,

Is there an equivalent parameter to AUTO_OPTIONS in ArduPlane? I am using v4.3.5 (official release) and there is no such option. Is there any way I can enable/disable arming in AUTO mode with ArduPlane v.4.3.5? Thanks!

rmackay9, 2023-07-08:
Hi @jankrystianserafim,  I'm pretty sure that Plane allows arming in Auto by default.. Rover does and it is actually originally a child of Plane.