ArduPilot/ardupilot #4924: Inconsistent time in mavlink GPS messages

BinhSw, 2016-10-03:
#### Issue details
I am noticing something strange when reading the telemetry data for GPS, and Attitude of the airplane. I am using the mavlink_global_position_int_t message for GPS data, and mavlink_attitude_t for attitude data. When I plot the time interval between the time_boot_ms of the GPS messages, it wasn't consistent between each message, but there was a pattern.
[image]
Does this have to do with how ardupilot schedules the sending of mavlink messages?

#### Platform
Ardu Plane

#### Airframe type
Fixed wing aircraft

#### Hardware type
Ardupilot Mega 2.6

magicrub, 2016-10-03:
Hi @BinhSw , thanks for taking the time to post this error you found. I see you're using APM 2.6 hardware which means you're using old firmware so this problem has most likely been fixed already. However, before I close this issue it would be nice for our GPS maintainer, @WickedShell, to chime in to confirm.

BinhSw, 2016-10-03:
Thanks for replying quickly!
I will probably upload the latest version of the software.
It would be great if they fixed this in the new version.

WickedShell, 2016-10-03:
@BinhSw I'd guess that this is just a telemetry scheduling issue. If you look in the dataflash logs do you see this same oscillation? (I'm guessing it will disappear)

BinhSw, 2016-10-03:
Sorry, I am new to the ardupilot.
How do you view the dataflash logs?

magicrub, 2016-10-03:
http://ardupilot.org/planner/docs/common-diagnosing-problems-using-logs.html

BinhSw, 2016-10-04:
I just uploaded the latest version of arduplane, and the problem seems to have been fixed.
I set the GPS telemetry rate to 10hz, and the time intervals between messages are consistently 200, 0, 200, 0, 200 milliseconds. Is this because the GPS only updates 5hz?
The attitude update intervals seem to be good too, 100 +- 3 ms.

magicrub, 2016-10-04:
Glad you confirmed it was fixed! The GPS update rate depends on the GPS, but yes 5 Hz is common. Closing...

closed 2016-10-04 by magicrub

WickedShell, 2016-10-04:
global_position_int != GPS postion for the record, so 10Hz is totally
reasonable. It's the position from the AHRS rather then the GPS.

BinhSw, 2016-10-04:
If it is the AHRS, why is the time interval between messages 200, then 0? Is the position from AHRS more accurate than from GPS?

WickedShell, 2016-10-04:
AHRS is the result from fusing in the EKF or DCM with the accelerometers.
(On a APM 2.6 you can't run the EKF so it's just the DCM).