ArduPilot/ardupilot #4926: Plane: resolve Error Terrain Alt Variance EKF warning message

CraigElder, 2016-10-04:
GCS indicates Error Terrain Alt Variance

Plane 3.7

#### Platform
[  ] All
[  ] Antenna Tracker
[  ] Copter
[ x] Plane
[  ] Rover

magicrub, 2016-10-04:
I think this has been reported a dozen times in various places. Solution
was to remove the gcs message from the ekf. Has that gone in?

closed 2016-10-04 by magicrub

OXINARF, 2016-10-04:
For reference: https://github.com/ArduPilot/ardupilot/commit/b6867e085d213f7d76f3df0c53def800bd6fc82e

R-Lefebvre, 2016-10-04:
Ok, but I think the larger issue is "Why is this happening"?  I am getting the error, even though I don't have a Lidar installed.

Removing the error reporting is just ignoring whatever the underlying issue is here.

CraigElder, 2016-10-04:
I think it is premature to close this issue. The patch is in master. It needs to be included in 3.7.1

OXINARF, 2016-10-04:
@CraigElder Once it is fixed in master, issues are closed. It would be a nightmare to close issues based on releases (specially because we have different releases for vehicles).

@R-Lefebvre If you don't have a Lidar nor Optical flow it doesn't seem unreasonable to me that the terrain alt variance would be big since you don't have a way to measure it, but only @priseborough can really confirm that. Also, the patch doesn't remove any error reporting, it simply only reports the variance when the EKF is using it.

R-Lefebvre, 2016-10-04:
As I understand it, the EKF is trying to judge the terrain based on difference between OF or Lidar, and what the terrain database is saying.

If there is no Lidar or OF attached, why is the EKF even attempting this?  What is it comparing to?

CraigElder, 2016-10-04:
@OXINARF We need to have this fix included in 3.7.1 and we need an open issue for it to appear here https://github.com/ArduPilot/ardupilot/milestone/24

OXINARF, 2016-10-04:
@CraigElder It appears there, just look in the closed issues. But that's not even a good way to manage things in GitHub because this issue applies to all vehicles, not only Plane.

@R-Lefebvre Have you sent the log to Paul?

priseborough, 2016-10-04:
@R-Lefebvre  the EKF is not using the terrain database - it uses assumptions about max terrain gradient though. Remember this particular estimator is primarily there for use when using optical flow or when using the baro/GPS at high altitude and range finder at low altitude mode.

magicrub, 2016-10-04:
Terrain is a keyword, I suggest we rephrase the message.

R-Lefebvre, 2016-10-05:
@priseborough So do you need logs to find the problem?  I'm getting it nearly every flight.  I've looked at my logs, and nothing seems to be logged in the dataflash logs?  No messages, errors or events.  I only have it in the TLOGs.