ArduPilot/ardupilot #4935: C'ant roll in FBWA-B mode

samys13, 2016-10-05:
#### Issue details
I can't roll or pitch my flying wing in fbwa or b mode. ailerons are just moving 2mm in roll and not move in pitch so can't roll and pitch as i want in fly.
i have set the nav_roll_cd at 4500.
i can easily move at 45° in manual and stabilize but not in all other mode.
does anyone know why please?

#### Version
arduplane 3.4
#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover

#### Airframe type
flying wing

#### Hardware type
mini apm 3.1

#### Logs

magicrub, 2016-10-05:
This is a support question, github issues is for developer bug reporting or new feature discussions. Please re-post your question at discuss.ardupilot.org

closed 2016-10-05 by magicrub

samys13, 2016-10-06:
hooo ok i'm sorry. thanks for the info