ArduPilot/ardupilot #4939: AHRS_Orientation configurations do not follow Tait-Bryan conventions

Georacer, 2016-10-07:
#### Issue details
Currently, the list of options provided for the `AHRS_Orientation` parameter do not follow the aerospace convention (Tait-Bryan) for Euler angles, according to which:
-180<yaw<=180
-90<pitch<=90
-90<roll<=90
It is also not clear to me if at least the YPR order is used, but I think this is the case, thankfully.

This results in confused users (e.g. http://discuss.ardupilot.org/t/ahrs-orientation-yaw-180-roll-90/11936/3) which are used to aerospace angles.

Perhaps a re-naming of the parameter options would clarify the use of AHRS_Orientation.

#### Version
Current

#### Platform
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
All

#### Hardware type
All

#### Logs
N/A

gmorph, 2016-10-10:
We can just update the description of each parameter value if that's going to make it better.  I'm not entirely convinced its needed however.  I don't think most users are "used" to aerospace angles.  Perhaps we should also include what the aerospace angle equivalent is.  However I'd suggest if they are used to aerospace angles they can probably work it out for themselves.

magicrub, 2016-10-10:
There's also only so much we can do for apm 2.5 users which is the case
with this linked discuss post. Documenting the params and such is pretty
much all that could and should be done.

gmorph, 2016-10-10:
The issue still affects non APM users i.e. current hardware.  AHRS_Orientation parameter is still exactly the same and its documentation is still exactly the same.

tridge, 2016-10-11:
the reason we didn't use those conventions is it gives negative numbers, and you can't have a C define for "YAW-90" as the minus isn't allowed. YAW270 is pretty clear I think. I actually think it is clearer than "YAW-90" as the - could be interpreted as a hyphen on reading

gmorph, 2016-10-11:
I agree that the simple ones like YAW270 etc are easy to understand.  It was the one cited in the example above:
Roll = 90 Because it's on it's right side facing forward.
Yaw = 180 Because the direction arrow is backwards
which mapped to "Roll270Pitch180"
that I really struggled to get my head around and thought that the aerospace angles might help.  
Note I wasn't thinking of changing any code - just having an extra description column and mapping the current list of settings to the aerospace angle equivalents.

Naterater, 2019-01-08:
I think the conventions are clearly documented and this issue should be closed.

slovak194, 2019-08-19:
> I think the conventions are clearly documented and this issue should be closed.

Hi, could you please clarify where it is documented?

IamPete1, 2021-08-07:
I think the conclusion is that yes there not matching the correct ranges, there also not in radians, but were not planning to change them, so I'm going to close. Not nearly as big a issue as it once was since we can no auto orientate compasses. 

Documentation is here: 
https://ardupilot.org/copter/docs/common-mounting-the-flight-controller.html#alternative-orientations

closed 2021-08-07 by IamPete1