ArduPilot/ardupilot #4950: safety switch not disabling  twin engine plane

yak-54, 2016-10-09:
Issue details

Ok i have twin engine plane running 3.7.0 and 3DR pixhawk
 i have throttle in one in RC3 and the other RC11 and RC11_FUNCTION: is set to 70 throttle all work's fine but i was testing power system with a watt meter then i finished pressed the safety switch  i hit the throttle by mistake and the plane took off into the wall the left throttle [ RC11 ] does not disarms pressing safety switch 

Version
plane 3.7.0

Platform
 Plane

Airframe 
Fixwing

3DR pixhawk

magicrub, 2016-10-09:
Thanks, Colin! Do you have a params file you can share so we can see your configuration?

yak-54, 2016-10-09:
yep can do all done

magicrub, 2016-10-10:
@tridge I can look into this for you

yak-54, 2016-10-11:
some else i have notice that none of the AUX channels get disabled as i do have 
left Ail in RC9
left Elev in RC10
left throttle in RC11
the resign why i did run Y leads on the servo's till i had one blown  the fuse on a pixhawk i lost the hole plane

yak-54, 2016-10-23:
not yet but i have found out that it does not disable any of the AUX channels there all still live

gmorph, 2016-12-31:
Any updates or can we close this out?

yak-54, 2016-12-31:
have not tested it yet will do to night they say its fixed the plane  just been hanging from the roof waiting for a fix :)

yak-54, 2016-12-31:
i will test it out tonight

yak-54, 2016-12-31:
just put 3.7.1 still the same does not disarms by pressing hardware switch and disarming in mission planer

WickedShell, 2017-05-29:
@tridge was this fixed as part of the SRVChan stuff?

tridge, 2017-05-29:
I believe this is fixed now, but I'd appreciate confirmation from @yak-54. Please test the latest build of master and see if its OK now.

yak-54, 2017-05-29:
Good will test it tonight when i get home from work :)

OXINARF, 2017-05-30:
@yak-54 Yes, latest Plane beta should work.

yak-54, 2017-05-30:
Yep all good thanks  safety switch is now disabling all servo output 
thank you all 
now just have to work out why i am getting bad log error

WickedShell, 2017-05-30:
Thanks for verifying it works correctly now! I'd post a topic on discuss if you can't resolve the bad logging warning yourself.

closed 2017-05-30 by WickedShell