ArduPilot/ardupilot #4952: No flightlog generated - LOG_BITMASK auto-changes to 0, and back to 65535 after reboot(s?).

Sulter, 2016-10-10:
#### Issue details
After a long flight there were no logfile generated on our fixed-wing platform. Fortunately I've got the .tlog from the ground station. 
It's worth noting that Mission Planner showed (and shows in the tlog) "Bad or no terrain data" message, which is the only thing out of the ordinary.

Looking at diffs between the params (generated with mavparams.py from the logs):
bin file generated before the flight (4.BIN): LOG_BITMASK, 65535
from the tlog doing the flight: LOG_BITMASK, 0
next bin file generated by the pixhawk (5.BIN): LOG_BITMASK,65535

Looking at the SYS_NUM_RESETS, I can see that there were 4 resets between the two 4.bin and 5.bin, but I don't remember how many times the plane was armed in between (but at least once, obviously!). There were no resets between 4.BIN and the flight (TLOG).

I've tried many different things but I haven't been able to reproduce the issue. And there are 30+ flights on both platforms with no similar issues.

A few months ago, I've also observed that there were no logfile generated after a flight on one of our  hexacopteter, unfortunately I didn't collect any other data from then (no tlog), but I just observed it.

#### Version
Arduplane 3.7.0
Arducopter 3.3.2
Mission Planner 1.3.41

#### Platform
[  ] All
[  ] AntennaTracker
[ x ] Copter
[ x ] Plane
[  ] Rover

#### Airframe type
Fixed-wing 
Hexacopter (DJI S1000)

#### Hardware type
3DR Pixhawk (one on each platform, purchased from different re-sellers at different time)
Verbatim 8GB SD card (one on each platform)

#### Logs
Params files (from 4.BIN, the flight/tlog and 5.BIN):
[params.zip](https://github.com/ArduPilot/ardupilot/files/519188/params.zip)

Logs (the 4.BIN, the flight/tlog and 5.BIN):
[logs.zip](https://github.com/ArduPilot/ardupilot/files/519189/logs.zip)

tridge, 2016-10-11:
this happens when the microSD card fails to mount properly or fails to initialise. The LOG_BITMASK is temporarily set to 0 to indicate to the logging code that it should not attempt to do any logging.
The "bad or no terrain data" is also caused by lack of a microSD card (as terrain data is stored on the microSD)

closed 2016-10-11 by tridge

skystarUK, 2016-11-03:
I have the same issue, no logs were saved, then I discovered the log_bitmask was at 0, so I checked through settings ago, and noticed it was back to the Pixhawk default...
Presumably based on tridge's response, I have a dodgy SD card?  I have ordered another anyway to give that a try but would be nice if we could get some more error messages to let us know the problems.