ArduPilot/ardupilot #4961: Initialization failed: Waiting for first HIL_STATE message

ruyi7952, 2016-10-11:
#### Issue details
Changed HIL_MODE 0 to 1 , reboot PixHawk 
Initialization failed: Waiting for first HIL_STATE message 
3.3/3.4 Successfully initialized,but Server does not work , No Aileron No Elevator No throttle No Rudder.....
#### Version
ArduPlane 3.5 3.6 3.7 and Git Master 
Mission Planner 1.3.39 1.3.41 and Git Master 
X-Plane 10.51r1 (BTW Old Versions like 10.45 10.50, GPS Location error in CGS)
[  ] All
[  ] AntennaTracker
[  ] Copter
[*] Plane
[  ] Rover

#### Airframe type
Plane Radio Control  Model  like PT-60  HiLstar17F Penguin

#### Hardware type
Pxhawk 2.4.7  (PXFMUv2)

#### Logs
_Please provide a link to any relevant logs that show the issue_

Waiting for first HIL_STATE message
Waiting for first HIL_STATE message
Waiting for first HIL_STATE message
Waiting for first HIL_STATE message
Initialising APM
Waiting for first HIL_STATE message
Waiting for first HIL_STATE message
Waiting for first HIL_STATE message
Waiting for first HIL_STATE message
Waiting for first HIL_STATE message
Initialising APM
Waiting for first HIL_STATE message
Waiting for first HIL_STATE message
Waiting for first HIL_STATE message
Waiting for first HIL_STATE message
Waiting for first HIL_STATE message
Initialising APM
Waiting for first HIL_STATE message
Waiting for first HIL_STATE message
Waiting for first HIL_STATE message
Waiting for first HIL_STATE message
Waiting for first HIL_STATE message
Initialising APM
Waiting for first HIL_STATE message
Waiting for first HIL_STATE message
Waiting for first HIL_STATE message
Waiting for first HIL_STATE message
Waiting for first HIL_STATE message
Initialising APM
Waiting for first HIL_STATE message
Waiting for first HIL_STATE message
Waiting for first HIL_STATE message
Waiting for first HIL_STATE message
Waiting for first HIL_STATE message
Initialising APM
Waiting for first HIL_STATE message
Waiting for first HIL_STATE message
Waiting for first HIL_STATE message
Waiting for first HIL_STATE message
Waiting for first HIL_STATE message
Initialising APM
Waiting for first HIL_STATE message
PX4v2 002E002F 33355108 33373538
PX4: 22b6f575 NuttX: d48fa307
ArduPlane V3.5.0 (379ca5ac)

ruyi7952, 2016-10-12:
v3.40 logs

Throttle armed!
command received: 
gps alt: 2324
init home

 Ready to FLY.
Initialising APM...
barometer calibration complete
Calibrating barometer
Waiting for first HIL_STATE message
Waiting for first HIL_STATE message
Waiting for first HIL_STATE message
Initialising APM...
Waiting for first HIL_STATE message
Waiting for first HIL_STATE message
Waiting for first HIL_STATE message
Waiting for first HIL_STATE message
Waiting for first HIL_STATE message
Initialising APM...
Waiting for first HIL_STATE message
Waiting for first HIL_STATE message
Waiting for first HIL_STATE message
Waiting for first HIL_STATE message
Waiting for first HIL_STATE message
Initialising APM...
Waiting for first HIL_STATE message
Waiting for first HIL_STATE message
Waiting for first HIL_STATE message
Waiting for first HIL_STATE message
PX4v2 002E002F 33355108 33373538
PX4: c4f39028 NuttX: d48fa307
ArduPlane V3.4.0 (3aa9eb0c)

magicrub, 2016-10-12:
Hil has been broken for a while, the latency is too high. Use sitl.

closed 2016-10-12 by magicrub

ruyi7952, 2016-10-12:
ok.I see. Will there be updates after Hil?

TheRibbonRed, 2017-08-09:
@ruyi7952 how did you manage to get it ready to fly? I'm still stuck with the repeated "Waiting for HIL_STATE message" and "Initializing APM" over and over like your first post. Whenever I arm it, it would show something like:
PreArm: Barometer not healthy
PreArm: Gyros not healthy
PreArm: Bad GPS pos
PreArm: Mixer error

I use the original (3DR) Pixhawk with ArduPlane V3.8.0, with various Mission Planner versions (including the newest v1.3) and QGCS (v3.2.2). I don't connect anything except the USB with the Pixhawk board, seeing as HIL/HITL is supposed to simulate the sensors through the board and get simulated outputs from the board.

Do I have to pass the Pre-Arm checking with an RC controller?