ArduPilot/ardupilot #4966: Add lxml and future to Windows ToolChain

rmackay9, 2016-10-12:
Recent changes to add dependencies on "lxml" and "future" python packages has broken the build for Windows.  We should add these packages to the windows toolchain:

Open this web page: http://www.lfd.uci.edu/~gohlke/pythonlibs/#lxml
search and download, "lxml-3.6.4-cp27-cp27m-win32.whl"
search and download, "future-0.15.2-py2-none-any.whl"

These can be installed by running:
pip install <filename>

SalahuddinAsh, 2016-10-17:
@rmackay9 I've tried to follow these instructions but when I run "pip install xxx" it gave me "xxx.whl is not a supported wheel on this platform". How can I fix it?

WickedShell, 2016-10-17:
> Windows installer here in case you are using pure windows :
> https://pypi.python.org/pypi/lxml/3.6.0

pip install xxx --user

It's on the sitl windows page.

magicrub, 2016-10-17:
pip install future works fine.

when I run "pip install lxml" it looks good for a while but fails when
compiling:

...
*fatal error: libxml/xpatch.h: No such file or directory*
*compilation terminated.*

*Could not find function xmlCheckVersion in library libxml2. Is libxml2
installed?*

running "pip install libxml2" doesn't help, gives me a not-found type error.

WickedShell, 2016-10-17:
@magicrub is this within cygwin? If so see the cygwin instructions, you
need libxml2-devel libxslt-devel python-devel and libexpat to get lxml to
install via pip

magicrub, 2016-10-17:
yup, that worked. Thanks Michael!

SalahuddinAsh, 2016-10-18:
Thanks guys for your replys.

@magicrub Excuse me, how did you installed these required packages?
I've tried running 
`$  python -m pip install --user libxml2-devel libxslt-devel python-devel libexpat`
But it gave me this error
```
You are using pip version 6.1.1, however version 8.1.2 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
Collecting libxml2-devel
  Could not find a version that satisfies the requirement libxml2-devel (from versions: )
  No matching distribution found for libxml2-devel
```
And when trying 
`$ pip install libxml2-devel`
It gave me this
```
Collecting libxml2-devel
  Could not find a version that satisfies the requirement libxml2-devel (from versions: )
No matching distribution found for libxml2-devel

```
I'm running these commands in Cygwin terminal

khancyr, 2016-10-18:
libxml2-devel should be install with Cygwin installer ! (maybe another way but this on works!)

magicrub, 2016-10-18:
cygwin installer

SalahuddinAsh, 2016-10-18:
Guys, I'm sorry but it doesn't work. I was trying the last hour to get it to work but it didn't. I even reinstalled Cygwin, installed the required packages, installed future and lxml but when I tried to build in eclipse it gave me the same error. 
`$ python -m pip install --user future
Collecting future
  Downloading future-0.15.2.tar.gz (1.6MB)
    100% |████████████████████████████████| 1.6MB 358kB/s
Installing collected packages: future
  Running setup.py install for future ... done
Successfully installed future-0.15.2
`
`$ python -m pip install --user lxml
Collecting lxml
  Using cached lxml-3.6.4.tar.gz
Installing collected packages: lxml
  Running setup.py install for lxml ... done
Successfully installed lxml-3.6.4
`
And when I built in eclipse it gave me this error
`c:/Users/Salahuddin/ardupilot/libraries/GCS_MAVLink/GCS_MAVLink.h:27:56: fatal error: include/mavlink/v2.0/ardupilotmega/version.h: No such file or directory
compilation terminated.
make[3]: *** [/c/Users/Salahuddin/ardupilot/ArduCopter/AP_Rally.cpp.o] Error 1
make[3]: Leaving directory `/c/Users/Salahuddin/ardupilot/modules/PX4Firmware/Build/px4fmu-v2_APM.build/c/Users/Salahuddin/ardupilot'
make[2]: *** [/c/Users/Salahuddin/ardupilot/modules/PX4Firmware/Build/px4fmu-v2_APM.build//c/Users/Salahuddin/ardupilot/module.pre.o] Error 2
make[2]: Leaving directory `/c/Users/Salahuddin/ardupilot/modules/PX4Firmware/Build/px4fmu-v2_APM.build'
make[1]: *** [/c/Users/Salahuddin/ardupilot/modules/PX4Firmware/Build/px4fmu-v2_APM.build/firmware.px4] Error 2
make[1]: Leaving directory `/c/Users/Salahuddin/ardupilot'
make: *** [px4-v2] Error 2

09:00:59 Build Finished (took 1m:8s.969ms)`

Any suggestions?

peterbarker, 2016-10-18:
On Tue, 18 Oct 2016, SalahuddinAsh wrote:

> And when I built in eclipse it gave me this error
> c:/Users/Salahuddin/ardupilot/libraries/GCS_MAVLink/GCS_MAVLink.h:27:56:
> fatal error: include/mavlink/v2.0/ardupilotmega/version.h: No such file or
> directory

Have  you run

git submodule update --recursive

?

SalahuddinAsh, 2016-10-18:
@peterbarker I've just done the following:
- uninstalled px4_toolchain and reinstalled the latest version from [here](firmware.ardupilot.org/Tools/PX4-tools/pixhawk_toolchain_installer_latest.exe) 
- delete ardupilot code and re-cloned it using
    - git clone https://github.com/ArduPilot/ardupilot.git
    - cd ardupilot
    - git submodule update --init --recursive
- Followed the instructions [here](http://ardupilot.org/dev/docs/editing-the-code-with-eclipse.html) to build in elcipse 
- Built using eclipse but also had the same error 
`c:/Users/Salahuddin/ardupilot/libraries/GCS_MAVLink/GCS_MAVLink.h:27:56: fatal error: include/mavlink/v2.0/ardupilotmega/version.h: No such file or directory`

What can I do else to solve it?

peterbarker, 2016-10-19:
@SalahuddinAsh Please try to build outside eclipse, using just "make" or "waf"

rmackay9, 2016-10-19:
Sid's produced a new Toolchain.  It would be great if @SalahuddinAsh could give it a try to see if it resolves the problems.  https://drive.google.com/file/d/0B9lyOylsxH2BbDhENzRXMmZBQm8/view

rmackay9, 2016-10-19:
The new toolchain has been uploaded to: http://firmware.ap.ardupilot.org/Tools/PX4-tools/ 	pixhawk_toolchain_installer_latest.exe

Hope this sorts out the problems.

SalahuddinAsh, 2016-10-19:
@peterbarker I've tried to run "make" from within px4 console but it didn't work. And I don't know how to run "waf" in Windows. Could you tell me?

@rmackay9 Many thanks for your reply. I really appreciate it.
- I uninstalled px4-toolchain and installed the latest version you provided and finally it built but with error. These are the last lines in console
`%% Generating /c/Users/Salahuddin/ardupilot/modules/PX4Firmware/Build/px4fmu-v2_APM.build/firmware.px4
fatal: Not a git repository: test/dspal_tester/cmake_hexagon/../../../../../../../../../.git/modules/modules/PX4Firmware/modules/src/lib/DriverFramework/modules/dspal/modules/test/dspal_tester/cmake_hexagon
fatal: 'git status --porcelain' failed in submodule dspal
fatal: 'git status --porcelain' failed in submodule src/lib/DriverFramework
make[2]: Leaving directory `/c/Users/Salahuddin/ardupilot/modules/PX4Firmware/Build/px4fmu-v2_APM.build'
%% Copying /c/Users/Salahuddin/ardupilot/modules/PX4Firmware/Images/px4fmu-v2_APM.px4
make[1]: Leaving directory `/c/Users/Salahuddin/ardupilot'
   text	   data	    bss	    dec	    hex	filename
 952028	   2848	  61196	1016072	  f8108	c:/Users/Salahuddin/ardupilot/modules/PX4Firmware/Build/px4fmu-v2_APM.build/firmware.elf
fatal: Not a git repository: c:/Users/Salahuddin/ardupilot/modules/../../.git/modules/modules/PX4Firmware
fatal: Not a git repository: c:/Users/Salahuddin/ardupilot/modules/PX4NuttX/nuttx//../../.git/modules/modules/PX4NuttX
Failed to get px4 hash
Failed to get nuttx hash
PX4 ArduCopter Firmware is in ArduCopter-v2.px4`
Is it normal? And if not, how can I solve it?

And in the problems tab there are some errors and warning of "xxx could not be resolved" 
like "Function 'isnan'", "Symbol 'is_arithmetic'", "Symbol 'value'", I think something is wrong with AP_MATH library, right?

- BTW, the second link is not working. It gave me this error
`You don't have permission to access /Tools/PX4-tools/pixhawk_toolchain_installer_latest.exe on this server.`

- When I've tried "waf" in linux it worked like magic  I think it's better than make. But as I couldn't navigate the code efficiently using (emacs + gtags) in linux I had to use eclipse and the only choice for eclipse (as I know) is the px4-toolchain in Windows, so I returned to Windows. so could you please tell me how to build using "waf" in Windows using console and eclipse (You can also add this to the wiki as there is nothing about it there)?

jpkh, 2016-10-19:
@SalahuddinAsh pixhawk toolchain error has been fixed on our system now. New file was missing permissions on satellite servers.

SalahuddinAsh, 2016-10-19:
@jpkh Do you mean that the error has been fixed in the tool chain that Randy linked recently? I've built using this tool-chain and the code was built successfully but still have the errors that I listed in my last comment. Or is there another thing that I have to do to remove the errors I have when building?
Thanks

rmackay9, 2016-10-19:
@SalahuddinAsh, it looks like maybe you need to run "git submodule init" and "git submodule update --recursive" from the ardupilot directory.

SalahuddinAsh, 2016-10-19:
@rmackay9 I've already done that yesterday and I also did it now but nothing changed :(

rmackay9, 2016-10-19:
@SalahuddinAsh, maybe you should join us on gitter.  https://gitter.im/ArduPilot/ardupilot

Maybe try deleting ardupilot/modules/PX4Firmware.. and maybe all other directory in /ardupilot/modules.

lvale, 2016-10-19:
I've tested with a clean Windows 10 install, and had no issues building with the latest toolchain

[image]

rmackay9, 2016-10-20:
@lvale, Thanks for testing!  I'm going to close this issue then.

closed 2016-10-20 by rmackay9

Embeddedkuba, 2017-02-21:
I have still the same error 'No module named future' (after instaling python and packs (lxml&future) ) Is it possible, that the version of Python is wrong? My versions: Python34

khancyr, 2017-02-21:
@Embeddedkuba please open a post on discuss about, detailling your config, what command you use and the error show ! 
Normally, it is working ..

rohts-patil, 2017-10-05:
@Embeddedkuba I am facing the same issue. After installing future I still get No module named future.
Have you found out a solution?

tuloski, 2017-12-21:
This is still an issue...

khancyr, 2017-12-21:
Install future for Python2 not Python3