ArduPilot/ardupilot #4969: Include error

suneelbelkhale, 2016-10-13:
#### Issue details
I get this error when running make for ArduCopter, but the others as well: 

> fatal error: include/mavlink/v2.0/ardupilotmega/version.h: No such file or directory
 #include "include/mavlink/v2.0/ardupilotmega/version.h"
                                                                                             ^
compilation terminated.

Also, running make by itself, as the documentation suggests, does not actually run make, since you have to specify the make target (px4-v1, etc.). Are apm boards supported? It doesn't look like make apm is an option...

#### Platform
[x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

rmackay9, 2016-10-13:
@suneelbelkhale,

You're compiling on windows?  If yes, I'm pretty sure the issue is the same as this: https://github.com/ArduPilot/ardupilot/issues/4966

We need to update the Windows Toolchain but until then you'll need to try and install the "future" and "lxml" packages.

suneelbelkhale, 2016-10-14:
Hi!
I'm actually on Mac, but I think it was just something about the way I installed APM planner. When I reinstalled, everything worked fine.

Thanks!

closed 2016-10-14 by suneelbelkhale