ArduPilot/ardupilot #4971: bug in AP_Baro_MS5611.cpp

night-ghost, 2016-10-13:
#### Issue details
there is a bug in AP_Baro_MS5611.cpp, which causes I2C bus lockups

in _init() it takes two semaphores: free semaphore from Util
`_sem = hal.util->new_semaphore();`

and bus semaphore from driver
`if (!_dev->get_semaphore()->take(10)) {`

But in _timer used only free semaphore!
`if (_sem->take(HAL_SEMAPHORE_BLOCK_FOREVER)) {`

which don't prevents bus conflicts.

moreover, _timer procedure works at interrupt level so it should use nonblocking semaphores.

Seems to me that this bug can be fixed  by using the bus semaphore only: change line to
`_sem = _dev->get_semaphore();`

to fix 2nd enough to increase frequency of _timer() procedure and try to take semaphore at beginning of it.

at least after these changes has been a half-hour no bus conflicts

gmorph, 2016-10-18:
Hmmmm. AP_Compass_BMM150.cpp seems to have the same code issue and so does DataFlash_File.cpp.  How/where are you seeing the bus conflicts?

Can you create a PR of your code changes?

Thanks, Grant.

night-ghost, 2016-10-18:
>How/where are you seeing the bus conflicts?
hardware debugging via SWD

>Can you create a PR of your code changes?
Only with RevoMini HAL because it to difficult to separate changes. But there is enough information in issue itself.

tridge, 2016-10-19:
well spotted! 
Do you want to do a PR or shall I fix this?

tridge, 2016-10-19:
ahh, I see grant already asked - no problem, I'll fix it. Looks easy to fix now you've found it.

tridge, 2016-10-19:
I've looked at it a bit more and there really should be two semaphores. One is protecting the data structure and the other protecting the bus. We don't want to use the bus one to protect the data structure as bus operations can take milliseconds and we don't want the main thread to be waiting on a bus operation to complete.
I've done a PR here: https://github.com/ArduPilot/ardupilot/pull/5026
could you please test this on your board?

tridge, 2016-10-19:
actually, I have now closed that PR. I realised it must be a bug in the RevoMini HAL.
If you look at HAL_Linux you'll see that the register_periodic_callback() sets up the callback on the per-bus thread. That uses start_cb() before calling each callback and that takes the device semaphore.
So I suspect the RevoMini hal is missing the take of the semaphore before it calls the callback.
ping to @lucasdemarchi

lucasdemarchi, 2016-10-20:
Yes. Lock is take by the bus periodic callback infra: https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_HAL/Device.h#L113

night-ghost, 2016-10-20:
> Lock is taken by the bus periodic callback infra

This is design flaw, as I see. Because there is some *independent* buses, and scheduler don't knows about which one used by given callback, it should prevent ANY operations, even on another buses.  Thus lost the advantage of independent buses. So the best place for semaphore is callback itself. 
It is also possible to use the bus semaphore at registration of callback, but this is a mixture of levels of responsibility.

night-ghost, 2016-10-20:
Moreover, Baro  driver yes, is called via this "periodic callback" - but compass don't use it!
```

void AP_Compass_HMC5843::accumulate()
{
    if (!_initialised) {
        // someone has tried to enable a compass for the first time
        // mid-flight .... we can't do that yet (especially as we won't
        // have the right orientation!)
        return;
    }

   uint32_t tnow = AP_HAL::micros();
   if (_accum_count != 0 && (tnow - _last_accum_time) < 13333) {
          // the compass gets new data at 75Hz
          return;
   }

   if (!_bus->get_semaphore()->take(1)) {
       // the bus is busy - try again later
       return;
   }
    
   bool result = _read_sample();
    
   _bus->get_semaphore()->give();
        
   if (!result) {
       return;
   }
```

So compass and baro uses the same i2c bus but different locking mechanisms. As a result, locking don't works.

night-ghost, 2016-10-20:
Heh, AP_Notify also uses I2C bus for Toshiba led outside of IO thread

tridge, 2016-10-20:
the callback is registered by a bus-specific function:
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Baro/AP_Baro_MS5611.cpp#L110
so it always uses the per-bus semaphore.

> So compass and baro uses the same i2c bus but different locking mechanisms. As a result, locking don't works. 

they do use the same semaphore. The compass uses _bus->get_semaphore() which returns the same semaphore object as the one used in the callback system used in the baro driver.

The reason for the inconsistency is we are in the process of moving to the new callback system in order to gain more bus parallelism. Not all drivers have been converted, but they should all still be working correctly. I don't yet see any actual bugs in what you've described.

tridge, 2016-10-20:
> Heh, AP_Notify also uses I2C bus for Toshiba led outside of IO thread

again, it is using the same per-bus semaphore correctly. It isn't using the same thread, but the semaphore does make it safe. As that driver uses to the new callback system it will gain efficiency as they all move to the same thread, but it won't change in correctness with respect to locking.

night-ghost, 2016-10-20:
Ok, as you wish.

closed 2016-10-20 by night-ghost

lucasdemarchi, 2016-10-20:
> This is design flaw, as I see. Because there is some independent buses, and scheduler don't knows about which one used by given callback, it should prevent ANY operations, even on another buses

Quiet the contrary: it does know about each  bus and there's a semaphore for each of them. Look at the Device class. Each device shares the same lock used in the bus it is in.  Particularly for MS5611, take a look on the last commits implementing this logic and you will see explanation of how much better it is now.

night-ghost, 2016-10-21:
>Quiet the contrary
The best solution is when all work with hardware does in single timer thread - initialization and LED too. Then there is no need to semaphores. To do this, we need one-time functions in scheduler or driver, which executes on the same IO thread.

PS, I think that the return value of the function should be enum, not bool: OK, FAILED, CANCEL. Then, the scheduler will know that the function failed to do necessary and it must be run in the next tick. See the same MS5611 - there is 2nd semaphore, and if it fails function can't do anything, only wait - but wait forever on interrupt level is a not good idea.