ArduPilot/ardupilot #4972: Feature request - new mode "Fast descent"

night-ghost, 2016-10-13:
it is known that helicopters and copters may not descent rapidly due to the formation of a vortex ring and loss of stability, known as "propwash". Therefore, the rate of descent in modes RTH and Land is rather small, and when flying at high altitude time of descent is the main component of time.
But there is a way to quickly reduce height - to descent like airplane. That is, the tilt copter to angle limit of Stabilize at low throttle, but also add some Yaw to avoid large bias. As a result, the copter in the air makes the coil, and the achievable rate of descent is several times higher than as usual. 

I propose to add a new mode "Fast descent" with an automatic exit from it at a height of 30 meters approximately to stabilize in level and to reduce the rate of descent.

rmackay9, 2016-10-13:
Yes, it's a good idea and it would look really cool coming down as well.  Leonard and I (and others) have discussed this in the past.  We were thinking the vehicle would descend in a spiral (but we weren't thinking it would adjust it's yaw, just roll and pitch).  It could be implemented as a new flight mode and it could be a fair bit like circle mode.

I'm happy to give advice, but this isn't something I could personally take on in the near future.

night-ghost, 2016-10-13:
I did not expect for rapid implementation - just let it be.

PS. Yes, there is another mode of even more rapid descent - to roll upside down and give a little throttle, allowing copter fall freely, and at an altitude of about 100 meters add throttle to full and perform a reverse rotation. But this method is not appropriate with the presence of gimbal. And one time, at the final stage, the speed was too high and the air flow breaks synchronization of ESC. Okay that copter was indestructible :)

rmackay9, 2016-10-13:
@night-ghost, yes, I've thought of the inverted fall thing as well, I think that would at least be incredibly cool to see even if it's perhaps a little to scary to get much practical use.  You know, really the implementation is not that difficult actually... everything is there in the attitude and position controller libraries.

night-ghost, 2016-10-13:
>everything is there in the attitude and position controller libraries.

Yes, I know, but to use a tool one must be learned to use it. I'm more on bare metal, 3d math for me is a dark forest.

ClementChristomanos, 2016-12-01:
Hi guys,
I would recommend not making it a flight mode, but a specific controller instead.
Point is, this feature could be used by other flight modes. For instance, LAND mode could use this feature and stop using this fast descent controller when it reaches a reasonable altitude (kind of what's done right know, without the spiral descent). Same for RTL. 
Another use case would be to have this feature activate when AUTO or GUIDED modes command descent phases with a sink rate greater than a given value. This would be really useful when flying from 300ft to 50ft for instance.

Having this feature would allow for greater flight times, as descent phases can be very long and thus battery consuming.

night-ghost, 2016-12-01:
Thanks, a very good idea!