ArduPilot/ardupilot #4974: call new defined class method function to get data will disconnected mission planner with pixhawk

CrazyPopLin, 2016-10-13:
#### Issue details
I write a c++ class in arducopter to get serial data from audino. And I tested through Mavproxy to show the data. It works. Then I called some funcation of this class in `control_loiter.cpp` to get data, want to change some loiter flight, like increase the climb rate. But when I enable loiter mode, the mission planner will immediately disconnected with pixhawk. 

Any suggestion will be appreciated. thanks.

#### Platform
 Copter-3.4-dev

#### Airframe type
quadcopter

#### Hardware type
pixhawk

CrazyPopLin, 2016-10-13:
for some reason,this code will disconnected mission planner
```
    if(lidar.getrange() < 100 || waterManager.getrange() >=500) {  
        target_climb_rate = get_pilot_desired_climb_rate(channel_throttle->control_in);
        hal.console->printf("radio\n");
    }
```
but this works fine.
```
    if(waterManager.getWaterStatus() < 100 || waterManager.getWaterStatus() >=500) {  
        target_climb_rate = get_pilot_desired_climb_rate(channel_throttle->control_in);  
    }  
```

does anyone know why?

closed 2016-10-13 by CrazyPopLin

magicrub, 2016-10-13:
You're spamming a debug port in a control system loop. You're probably
generating data faster than you can transmit it.