ArduPilot/ardupilot #4980: Can't compile in linux ardupilotmega/version.h: No such file or directory??

SalahuddinAsh, 2016-10-13:
#### Issue details
Hi everyone, 
I'm trying to build ArduCopter in linux (Ubuntu 16.04) 
I've followed the steps on the wiki [http://ardupilot.org/dev/docs/building-px4-for-linux-with-make.html](here)
When running "make px4-v2" in terminal, first time it worked, now it gives an error 
`fatal error: include/mavlink/v2.0/ardupilotmega/version.h: No such file or directory
 #include "include/mavlink/v2.0/ardupilotmega/version.h"
                                                        ^
compilation terminated.`
I've tried to delete the code from my pc and clone it from git again but didn't work too

#### Version
Master branch

#### Platform
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover

#### Airframe type
Y6

#### Hardware type
pixhawk

#### Logs
_Please provide a link to any relevant logs that show the issue_

magicrub, 2016-10-13:
make is about to be depreciated for linux in favor of waf, however the there's only a single header on the wiki page that mentions the directions are located elsewhere. The wiki needs to show the waf build procedure.

Please try waf and report back: https://github.com/ArduPilot/ardupilot/blob/master/BUILD.md

rmackay9, 2016-10-14:
I think it's probably related to the new requirement for "future" and "lxml".  I think a "pip install future" and "pip instal lxml" will sort it out.  I believe the "prerequisite" scripts in the Tools directory have now been updated to install these packages.

ShingoMatsuura, 2016-10-14:
Thanks! @rmackay9. I could build normally after "pip install future" and "pip install lxml".

SalahuddinAsh, 2016-10-14:
Thanks guys for your help. I really appreciate it :)
@magicrub  I've tried to install waf in Linux following this instruction https://waf.io/book/ but couldn't do that. Any suggestion?
@rmackay9 It worked great. Althought when I tried to "pip install lxml" it gave me error
`Failed building wheel for lxml
  Running setup.py clean for lxml
Failed to build lxml
Installing collected packages: lxml
  Running setup.py install for lxml ... error
...
error: command 'i686-linux-gnu-gcc' failed with exit status 1`
But when I tried to build ardupilot using `make px4-v2` it worked. 

When I tried to build ardupilot in windows using px4-toolchain eclipse, it gave me the same error 
` c:/Users/Salahuddin/ardupilot/libraries/GCS_MAVLink/GCS_MAVLink.h:27:56: fatal error:
include /mavlink/v2.0/ardupilotmega/version.h: No such file or directory` Although I've cloned the latest code from github and also with `git submodule init update`. What do you think?

Thanks again for your time

lucasdemarchi, 2016-10-14:
You don't need to install waf. It's already in our repository. You just
need to follow what's in the build instructions @magicrub linked

SalahuddinAsh, 2016-10-14:
@lucasdemarchi Thanks for your time. If it is in the repository, then why when I run `./waf configure --board px4-v2` within ardupilot/ArduCopter directory it gives me an error `bash: ./waf: No such file or directory`?
When I tried to google, someone said that ns-3 is needed so I followed the instructions in this link but it didn't work as well!!
What should I do to get waf to work successfully? 
Many thanks

lucasdemarchi, 2016-10-14:
Are you in the master branch?  How did you clone the repository?

SalahuddinAsh, 2016-10-14:
```
git clone https://github.com/ArduPilot/ardupilot.git
cd ardupilot/ArduCopter
git submodule init
git submodule update
```

lucasdemarchi, 2016-10-14:
Something went wrong with your clone I imagine.... The file is there:
https://github.com/ArduPilot/ardupilot/blob/master/waf

SalahuddinAsh, 2016-10-14:
I already have the file too in ardupilot directory. I tried running ./waf while being in ardupilot directory ( and not in ArduCopter as I did before) and it's working now. Thank you very much for your help.

Is there some way to integrate it with eclipse in windows as using make gives me many errors and warning? or it's only available in linux? 

I'm sorry, but one more question. What is the best development environment for ardupilot development. I'm facing troubles with eclipse and QT in windows. And in linux I can build well now but not Ok with reading, navigating, and editing the code. I really appreciate your help.

Thanks

lucasdemarchi, 2016-10-14:
> Is there some way to integrate it with eclipse in windows as using make gives me many errors and warning? or it's only available in linux?

It's possible to integrate with eclipse, you'd need to just tell it to run these commands to compile... most of people developing ardupilot are using Linux, though. Maybe @guludo or @rmackay9 can help you. 

> I'm sorry, but one more question. What is the best development environment for ardupilot development. I'm facing troubles with eclipse and QT in windows. And in linux I can build well now but not Ok with reading, navigating, and editing the code. I really appreciate your help.

I've seen a variety of IDE/editors being used. I myself use neovim. I've seen others with emacs, eclipse, sublime, qtcreator... People tend to use whatever editor they are used to from other projects.

SalahuddinAsh, 2016-10-14:
Many thanks for your time @lucasdemarchi, you have no idea how I appreciate your help. 
I'm trying to use 
- emacs and waf on linux for editing and compiling  
- MAVProxy in linux for simulation which communicate with Mission Planner in Windows for simulation

Thanks again

closed 2016-10-15 by lucasdemarchi

magicrub, 2016-10-15:
@SalahuddinAsh do you suggest any changes to the documentation that would be more clear and could have avoided this confusion?

SalahuddinAsh, 2016-10-15:
@magicrub yes. I would like to suggest couple of editing to the wiki as from the experience I went through as a newbie.

I think it would be very helpful to update the documentation related to
- building the code 
- IDEs for navigating and editing the code

I just knew about waf from your first comment here. In wiki  there is just a hint about waf and the main focus is on building using make. I just noticed it when I google waf in the wiki after your comment. So I think it will be meaningful to add it in the left navigation menu in the wiki as the recommended approach to compile the code in linux. And also in Windows by showing in details how to integrate it with eclipse.

The second point is navigating and editing the code. The wiki documentation about this is really very poor, at least from my point of view. I've spent long time to try to get eclipse and qt to work in Windows but it was painful for me. And I also tried to get Visual Studio to work but I figured out, may be I'm wrong, that it only support old versions of ardupilot which has .pde extension instead of .cpp. And now I'm trying to learn and configure emacs in linux so that I can navigate easily through the code base to learn it. 
It would be great if someone added some documentation about this part in the wiki (i.e., how, in details, to configure emacs, and other IDEs,  to work nice with ardupilot). I think this will save a lot of effort for the newbies and also for the experts trying to answer the newbies' question.

Please tell me what do you think about these two points and sorry for my long answer and thanks for your time.

lucasdemarchi, 2016-10-15:
@SalahuddinAsh @magicrub there's an issue open in the wiki for that: https://github.com/ArduPilot/ardupilot_wiki/issues/565

SalahuddinAsh, 2016-10-15:
Thanks for taking it seriously. Someone mentioned it is important but not a priority. This may be true for experts but I think it's a high-priority issue for newbie. Thanks again for your help.

SalahuddinAsh, 2016-10-18:
Could anyone please tell me how can I tell eclipse to build using **waf** as I'm facing troubles building using make in windows. https://github.com/ArduPilot/ardupilot/issues/4966