ArduPilot/ardupilot #4986: Using ROS/Gazebo Simulator with SITL （catkin_make err)

kiti1013, 2016-10-14:
#### Issue details
When I follow the website guide（http://ardupilot.org/dev/docs/using-rosgazebo-simulator-with-sitl.html） to try to compile ros package, the emergence of the above situation
#### Version
ubuntu 14.04
ros indigo
#### Platform
[  ] All

#### Logs
`atera@atera-Firebat-F640:~/workspace/rosworkspace$ catkin_make
Base path: /home/atera/workspace/rosworkspace
Source space: /home/atera/workspace/rosworkspace/src
Build space: /home/atera/workspace/rosworkspace/build
Devel space: /home/atera/workspace/rosworkspace/devel
Install space: /home/atera/workspace/rosworkspace/install
####
#### Running command: "make cmake_check_build_system" in "/home/atera/workspace/rosworkspace/build"
####
####
#### Running command: "make -j8 -l8" in "/home/atera/workspace/rosworkspace/build"
####
[  3%] [  3%] Built target gflags_src
[  6%] Built target std_msgs_generate_messages_py
Built target glog_src
[  6%] Built target geometry_msgs_generate_messages_py
[  6%] Built target _mav_msgs_generate_messages_check_deps_CommandRateThrust
[  6%] Building CXX object mavros/libmavconn/CMakeFiles/mavconn.dir/src/interface.cpp.o
[  6%] Built target _mav_msgs_generate_messages_check_deps_CommandMotorSpeed
[  6%] Built target _mav_msgs_generate_messages_check_deps_MotorSpeed
[  6%] Built target std_msgs_generate_messages_lisp
[  6%] Built target _mav_msgs_generate_messages_check_deps_CommandTrajectory
[  6%] Built target geometry_msgs_generate_messages_lisp
[  6%] [  6%] Built target _mav_msgs_generate_messages_check_deps_CommandAttitudeThrust
Built target geometry_msgs_generate_messages_cpp
[  6%] Built target std_msgs_generate_messages_cpp
[  6%] Built target _mav_msgs_generate_messages_check_deps_CommandRollPitchYawrateThrust
[  6%] Built target _mav_msgs_generate_messages_check_deps_CommandVelocityTrajectory
[  6%] Built target _mavros_msgs_generate_messages_check_deps_ExtendedState
[  6%] Built target _mavros_msgs_generate_messages_check_deps_CommandInt
[  6%] [  6%] Built target _mavros_msgs_generate_messages_check_deps_FileRemove
Built target _mavros_msgs_generate_messages_check_deps_CommandTOL
[  6%] [  6%] Built target _mavros_msgs_generate_messages_check_deps_ActuatorControl
[  6%] Built target _mavros_msgs_generate_messages_check_deps_WaypointList
Built target _mavros_msgs_generate_messages_check_deps_FileTruncate
[  6%] [  6%] Built target _mavros_msgs_generate_messages_check_deps_CommandLong
Built target _mavros_msgs_generate_messages_check_deps_Vibration
[  6%] Built target _mavros_msgs_generate_messages_check_deps_State
[  6%] Built target _mavros_msgs_generate_messages_check_deps_RCOut
[  6%] Built target _mavros_msgs_generate_messages_check_deps_FileClose
[  6%] Built target _mavros_msgs_generate_messages_check_deps_FileRename
[  6%] Built target _mavros_msgs_generate_messages_check_deps_GlobalPositionTarget
[  6%] Built target _mavros_msgs_generate_messages_check_deps_WaypointClear
[  6%] [  6%] [  6%] Built target _mavros_msgs_generate_messages_check_deps_WaypointSetCurrent
Built target _mavros_msgs_generate_messages_check_deps_PositionTarget
Built target _mavros_msgs_generate_messages_check_deps_FileChecksum
[  6%] Built target _mavros_msgs_generate_messages_check_deps_FileEntry
[  6%] Built target _mavros_msgs_generate_messages_check_deps_RadioStatus
[  6%] Built target _mavros_msgs_generate_messages_check_deps_VFR_HUD
[  6%] Built target _mavros_msgs_generate_messages_check_deps_FileList
[  6%] Built target _mavros_msgs_generate_messages_check_deps_AttitudeTarget
[  6%] [  6%] [  6%] Built target _mavros_msgs_generate_messages_check_deps_ParamGet
Built target _mavros_msgs_generate_messages_check_deps_OverrideRCIn
Built target _mavros_msgs_generate_messages_check_deps_SetMode
[  6%] Built target _mavros_msgs_generate_messages_check_deps_BatteryStatus
[  6%] Built target _mavros_msgs_generate_messages_check_deps_FileMakeDir
[  6%] Built target _mavros_msgs_generate_messages_check_deps_RCIn
[  6%] Built target _mavros_msgs_generate_messages_check_deps_StreamRate
[  6%] [  6%] [  6%] Built target _mavros_msgs_generate_messages_check_deps_FileOpen
Built target _mavros_msgs_generate_messages_check_deps_CommandHome
Built target _mavros_msgs_generate_messages_check_deps_HomePosition
[... 85 lines cut ...]
[ 78%] [ 78%] Built target rotors_gazebo_bag_plugin
[ 79%] Built target rotors_gazebo_motor_model
Built target rotors_gazebo_multirotor_base_plugin
[ 79%] Built target rotors_gazebo_octomap_plugin
[ 79%] Built target rotors_gazebo_odometry_plugin
[ 80%] [ 80%] Built target rotors_gazebo_wind_plugin
Built target rotors_gazebo_sonar_plugin
/home/atera/workspace/rosworkspace/src/mavros/libmavconn/src/interface.cpp: In member function ‘mavconn::MsgBuffer* mavconn::MAVConnInterface::new_msgbuffer(const mavlink_message_t*, uint8_t, uint8_t)’:
/home/atera/workspace/rosworkspace/src/mavros/libmavconn/src/interface.cpp:89:28: error: too few arguments to function ‘uint16_t mavlink_finalize_message_chan(mavlink_message_t*, uint8_t, uint8_t, uint8_t, uint8_t, uint8_t, uint8_t)’
     mavlink_crcs[msg.msgid]);
                            ^
In file included from /opt/ros/indigo/include/mavlink/v1.0/ardupilotmega/../protocol.h:85:0,
                 from /opt/ros/indigo/include/mavlink/v1.0/ardupilotmega/ardupilotmega.h:30,
                 from /opt/ros/indigo/include/mavlink/v1.0/ardupilotmega/mavlink.h:33,
                 from /home/atera/workspace/rosworkspace/src/mavros/libmavconn/include/mavconn/mavlink_dialect.h:57,
                 from /home/atera/workspace/rosworkspace/src/mavros/libmavconn/include/mavconn/interface.h:38,
                 from /home/atera/workspace/rosworkspace/src/mavros/libmavconn/src/interface.cpp:22:
/opt/ros/indigo/include/mavlink/v1.0/ardupilotmega/../mavlink_helpers.h:68:25: note: declared here
 MAVLINK_HELPER uint16_t mavlink_finalize_message_chan(mavlink_message_t* msg, uint8_t system_id, uint8_t component_id, 
                         ^
make[2]: *** [mavros/libmavconn/CMakeFiles/mavconn.dir/src/interface.cpp.o] Error 1
make[1]: *** [mavros/libmavconn/CMakeFiles/mavconn.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j8 -l8" failed
`

khancyr, 2016-10-14:
Hi, we are currently update gazebo plugin. The wiki is outdated on this part.
You can find the updated documentation here https://github.com/khancyr/ardupilot_wiki/commit/1cb46959a1ed0c3994872c55ea2cb4f72615778f

squilter, 2016-10-29:
Wiki will be fixed soon. See ardupilot/ardupilot_wiki#572

closed 2016-10-29 by squilter