ArduPilot/ardupilot #4988: Can't compile in Windows Make

UWylnahn, 2016-10-14:
#### Issue details
Hi, I'm trying to compile ArduPlane firmware using instructions on http://ardupilot.org/dev/docs/building-px4-with-make.html and am running into some issues.

I copy-pasted the errors it gave here.
#### Logs
make[4]: *** [/c/users/(myname)/documents/github/ardupilot/ArduPlane/ArduPlane.cpp.o] Error 1
make[4]: Leaving directory `/c/users/(myname)/documents/github/ardupilot/modules/PX4Firmware/Build/px4fmu-v1_APM.build/c/users/(myname)/documents/github/ardupilot'
make[3]: *** [/c/users/(myname)/documents/github/ardupilot/modules/PX4Firmware/Build/px4fmu-v1_APM.build//c/users/(myname)/documents/github/ardupilot/module.pre.o] Error 2
make[3]: Leaving directory `/c/users/(myname)/documents/github/ardupilot/modules/PX4Firmware/Build/px4fmu-v1_APM.build'
make[2]: *** [/c/users/(myname)/documents/github/ardupilot/modules/PX4Firmware/Build/px4fmu-v1_APM.build/firmware.px4] Error 2
make[2]: Leaving directory `/c/users/(myname)/documents/github/ardupilot'
make[1]: *** [px4-v1] Error 2
make[1]: Leaving directory `/c/users/(myname)/documents/github/ardupilot/ArduPlane'
make: *** [px4] Error 2

I figured it might be because my submodules aren't updating correctly, so I ran the command to update them and I'm getting the errors
Failed to recurse into submodule path 'modules/PX4Firmware'

I'm not sure what's going on, or if it's an error on my end, but I successfully compiled everything 2 weeks ago.

magicrub, 2016-10-15:
We've been having a lot of build failure reports in the last week or two, I think something has changed and hopefully its just a matter of updating the wiki.

That said, I see you're compiling px4-v1 where you probably meant px4-v2. Please confirm.

xshting, 2016-10-15:
me ,too

rmackay9, 2016-10-17:
Please try a "git submodule init" and "git submodule update --recursive".  If that doesn't work, then also try deleting the /ardupilot/modules/PX4Firmware and /ardupilot/modules/mavlink and run the two commands again.
The open issue is related to the adding of two new python modules, "lxml" and "future".  There's already an issue opened for this issue so closing this one.
https://github.com/ArduPilot/ardupilot/issues/4966

closed 2016-10-17 by rmackay9

xshting, 2016-10-18:
I have deleted the two directories then run the two commands ,it Can't compile yet:
[image]

magicrub, 2016-10-18:
easiest solution is to delete those folders and do init and update again,
they'll repopulate with correct paths.

khancyr, 2016-10-18:
@xshting as you delete the two directories, you need to reinit them again ! Use "git submodule update --init --recursive" that way it will update all submodule, init them if needed, and do the same for submodules submodules !