ArduPilot/ardupilot #4997: Plane: altitude.cpp and tecs use different altitudes

KIrill-ka, 2016-10-15:
#### Issue details
altitude.cpp use gps biased relative altutude. 
```
 ArduPlane.cpp: ahrs.get_position(current_loc)
 altitude.cpp: relative_altitude = current_loc.alt - home.alt
```

TECS uses pressure biased altitude:

```
TECS.cpp: _ahrs.get_relative_position_NED(posned)
         _height = - posned.z;

```
For example , in FBWB mode every time I release elevator stick, plane attempted to climb 7 meters.
That altitude difference I found in log (NTUN.AltErr)

#### Version
3.7.0

#### Platform
[ *] Plane

#### Hardware type
PX4

magicrub, 2016-10-16:
There's an ongoing battle with how altitudes are determined but we should only be looking at one at any given time. Seems you found a spot where it's mixed, thanks! We'll take a look. We have 6 sources of altitude:
- GPS
- baro
- rangefinder (lidar/sonar)
- optical flow (assisted /w a dedicated rangefinder)
- terrain database (offset by any one of the above)

tridge, 2016-10-18:
under what conditions do you see this? Is it when you've moved home? 
I'd like enough info to be able to reproduce this in SITL

tridge, 2016-10-18:
a DF log would be good too!

KIrill-ka, 2016-10-18:
> a DF log would be good too!

I don't want to publish complete log. But I can send it directly to you by e-mail.

> under what conditions do you see this? Is it when you've moved home? 

Nothing special. I didn't move home. Just GPS altitude drifted 7 meters up after ARMing.

KIrill-ka, 2016-10-18:
Log extract:

```

Just before ARM
ORGN, 66811328, 1, ... , 174.50999999999999
POS, 66812204, ..., 173.66999799999999, -0.83539706499999999
AHR2, 66812181, 2.96, 1.27, 203.71000000000001, 174.959991, ....
BARO, 66911769, -0.211727783, 101177.42999999999, 12.35, -0.73926246200000001, 66911, 0
BAR2, 66911769, 0.43378377000000001, 101088.67200000001, 11.59, -0.73926246200000001, 66911, 0

Before take-off:
AHR2, 129691974, 2.8700000000000001, -2.6299999999999999, 263.45999999999998, 175.91000399999999, ...
POS, 129691997, ..., 180.84999099999999, 0.97733300899999997
BARO, 129710997, 1.37106657, 101158, 13.69, -0.54133051600000004, 129710, 0
BAR2, 129710997, 1.78161192, 101072.141, 12.65, -0.54133051600000004, 129710, 0

mid flight:
AHR2, 541814385, 2.4700000000000002, 0.72999999999999998, 186.44, 219.81999200000001, ...
POS, 541814408, ...., 227.209991, 45.337097200000002
BARO, 541913583, 45.673816700000003, 100615.42200000001, 15.26, -0.90210676199999995, 541913, 0
BAR2, 541913583, 45.511146500000002, 100537.008, 12.77, -0.90210676199999995, 541913, 0
TECS, 541913614, 45.278625499999997, -0.46494779000000003, 41.959945699999999, ...
GPS, 542051147, 3, 300757600, 1918, 20, 0.56000000000000005, ..., 225.34999999999999, 11.399999599999999, 180, 0.44999998800000002, 1

landed, then disarmed:
ORGN, 1152911151, 0, ..., 182.41999999999999
BARO, 1152912746, -0.040882688, 101156, 15.42, 0.077646210800000004, 1152912, 0
BAR2, 1152912746, -0.056213695600000002, 101086.836, 13.15, 0.077646210800000004, 1152912, 0
AHR2, 1152931869, -1.0900000000000001, -10.1, 27.98, 182.36000100000001, ...
POS, 1152931897, ..., 182.28999300000001, -0.124461479

```
ORGN didn't change during flight.
From "mid flight" messages we can see that all relative altitudes match (TECS, BARO, BARO2, POS = 45m)
AHR2 altitude = HOME altitude + BARO altutude  = 175+45 = 220
POS altitude = GPS altitude
Relative altitude used by altitude.cpp is POS altitude - HOME altitude = 227-175 = 52.
That's what TECS sees as input (it's not logged).
From the log I see that POS altitude generally follows GPS altitude.

magicrub, 2016-12-06:
This was discussed today in the dev call. We'll be looking into it

WickedShell, 2017-05-29:
Was addressed as part of https://github.com/ArduPilot/ardupilot/pull/5647

closed 2017-05-29 by WickedShell