ArduPilot/ardupilot #5007: Manual start position not working on 1.0.0

zverugaz, 2016-10-17:
#### Issue details
When I not connenct a GPS to my Pixhawk with Antenna Tracker firmware and set START_LONG/START_LAT to some position (40 and 40 for example) a AT icon still show on map on position 0 for latitude and 0 for longitude (left from Africa). When I install a 0.8 version everything work well - AT show himself on correct postition.

#### Version
1.0.0

#### Platform
[  ] All
[ X ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
Octa

#### Hardware type
pixhawk

#### Logs
https://www.dropbox.com/s/5ogze7lzfgknycf/2016-10-17%2020-26-42.tlog?dl=0

Naterater, 2019-01-08:
Is this still an issue with any more recent firmwares?

rmackay9, 2019-01-09:
I don't know, we'd need to investigate probably as part of a new Tracker beta period and release.

IamPete1, 2019-02-05:
looks like this issue is still present, I will see what I can do

IamPete1, 2019-05-06:
fixed

closed 2019-05-06 by IamPete1