ArduPilot/ardupilot #5010: Copter 3.4rc6 breaks compatibility with OSD

night-ghost, 2016-10-18:
[This commit](https://github.com/ArduPilot/ardupilot/commit/412dca1615329cd593992aed99cbcc59f1462f2b)  removed RC_CHANNELS_RAW message without any graceful time and even without a mention in the ReleaseNotes. This is only message that used by OSD to get RSSI and channels (see [here](https://github.com/diydrones/MinimOSD-Extra/blob/master/MinimOsd-Extra/MAVLink.ino)). The same message is used in PlayUAV OSD.

So ALL owners of OSD boards will be very surprised with the release of 3.4

magicrub, 2016-10-18:
this sounds bad

night-ghost, 2016-10-18:
starting this commit https://github.com/ArduPilot/ardupilot/commit/53260969bb00ca30c11d823b84d1cb74e627128d

WickedShell, 2016-10-18:
> starting this commit 5326096

Argh! Breaks my stuff as well. This is really surprising since the MAVLink
XML never marked RC_CHANNELS_RAW as deprecated (And I didn't see any
comments in the code that this was deprecated). And it still doesn't show
that...

night-ghost, 2016-10-18:
May be it deprecated in Mavlink 2.0 but before device don't confirm its usage FC should maintain compatibility.

tridge, 2016-10-18:
fixed in master, it now sends RC_CHANNELS_RAW as well if using MAVLink1

closed 2016-10-18 by tridge

night-ghost, 2016-10-20:
breaking changes between RC5 and RC6 - it's cool!

Thanks for fast fixing.