ArduPilot/ardupilot #5019: Copter: compilation failed

night-ghost, 2016-10-19:
#### Issue details

make[3]: Entering directory `/mnt/disk_d/src/quad-copter/ardu-copter/3.4/ardupilot/modules/PX4Firmware/Build/px4fmu-v2_APM.build/mnt/disk_d/src/quad-copter/ardu-copter/3.4/ardupilot'
CXX:     /mnt/disk_d/src/quad-copter/ardu-copter/3.4/ardupilot/ArduCopter/control_guided_nogps.cpp
In file included from /mnt/disk_d/src/quad-copter/ardu-copter/3.4/ardupilot/libraries/GCS_MAVLink/GCS.h:10:0,
                 from /mnt/disk_d/src/quad-copter/ardu-copter/3.4/ardupilot/ArduCopter/Copter.h:39,
                 from /mnt/disk_d/src/quad-copter/ardu-copter/3.4/ardupilot/ArduCopter/control_guided_nogps.cpp:3:
/mnt/disk_d/src/quad-copter/ardu-copter/3.4/ardupilot/libraries/GCS_MAVLink/GCS_MAVLink.h:27:56: fatal error: include/mavlink/v2.0/ardupilotmega/version.h: No such file or directory
 #include "include/mavlink/v2.0/ardupilotmega/version.h"
                                                        ^
compilation **terminated.**

all found "version.h" files are:
APMrover2/version.h   
AntennaTracker/version.h
ArduCopter/version.h
ArduPlane/version.h
modules/PX4Firmware/Build/p~rt/include/nuttx/version.h
modules/PX4Firmware/Build/p~rt/include/nuttx/version.h
modules/PX4Firmware/mavlink~link/v1.0/ASLUAV/version.h
modules/PX4Firmware/mavlink~.0/ardupilotmega/version.h
modules/PX4Firmware/mavlink~nk/v1.0/autoquad/version.h
modules/PX4Firmware/mavlink~link/v1.0/common/version.h
modules/PX4Firmware/mavlink~v1.0/matrixpilot/version.h
modules/PX4Firmware/mavlink~ink/v1.0/minimal/version.h
modules/PX4Firmware/mavlink~vlink/v1.0/slugs/version.h
modules/PX4Firmware/mavlink~avlink/v1.0/test/version.h
modules/PX4Firmware/mavlink~nk/v1.0/ualberta/version.h
modules/PX4Firmware/src/lib/version/version.h
modules/PX4NuttX/nuttx/include/nuttx/version.h
modules/mavlink/pymavlink/g~nclude_v0.9/test/version.h
modules/mavlink/pymavlink/g~nclude_v1.0/test/version.h

#### Version
today's git clone.  The clone at 4e017bf5b3da4f2a9ffc2e1cc0a37b94edac2bdc compiles successfully

#### Platform
FedoraCore Linux

#### Hardware type
command was 
---
git clone [URL]
cd ardupilot
git submodule init && git submodule update
cd ArduCopter
make px4-v2

OXINARF, 2016-10-19:
That error means your MAVLink headers haven't been generated. Have you installed Python's future and lxml? You can do it with ``pip install future lxml`.

Since this isn't a bug in our end (it is a support question and those should be posted at the forum (http://discuss.ardupilot.org)), I'm closing this.

closed 2016-10-19 by OXINARF

rmackay9, 2016-10-20:
The new windows toolchain is here in case it's helpful:
http://firmware.ap.ardupilot.org/Tools/PX4-tools/pixhawk_toolchain_installer_latest.exe

night-ghost, 2016-10-20:
> Have you installed Python's future and lxml? You can do it with `pip install future lxml.

There should be check for this somewhere in Makefile, and an error should be generated when creating of headers failed, not at compile time. 

> The new windows toolchain

>Platform: FedoraCore Linux

You must be kidding...

peterbarker, 2016-10-20:
On Wed, 19 Oct 2016, night-ghost wrote:

>       Have you installed Python's future and lxml? You can do it with
>       `pip install future lxml.
> 
> There should be check for this somewhere in Makefile, and an error should be
> generated when creating of headers failed, not at compile time.

Sadly, the header generation is explicitly allowed to fail because the 
Python interpreter was crashing on exit (after successfully generating the 
headers) on Windows.

https://github.com/ArduPilot/ardupilot/blob/master/mk/mavgen.mk#L12

The waf build is a bit smarter; since the crash is a signal, it allows 
failure of python due to a signal but not due to a positive error code. 
Shoe-horning that into a Makefile could make for an unpleasant pastime...

night-ghost, 2016-10-20:
>Sadly, the header generation is explicitly allowed to fail because the Python interpreter was crashing on exit

but you can simply verify the existence of the file, for example version.h!

night-ghost, 2016-10-20:
>Have you installed Python's future and lxml? You can do it with `pip install future lxml.

Installing 'future' helps, thanks. BTW, is there a changelog where one can see changes of requirements?

OXINARF, 2016-10-20:
> is there a changelog where one can see changes of requirements?

No. And let me clarify that these new requirements are from MAVLink, not directly from ArduPilot.

flyflyfly001, 2016-11-07:
Hi, Have you solved the problem? I have meet the same problem as you. Could you give me some suggestion? Thanks a lot.

khancyr, 2016-11-07:
@flyflyfly001 sudo pip install future lxml then sudo pip install -U pymavlink mavproxy

flyflyfly001, 2016-11-07:
@khancyr OK, thank you khancyr. I will try it soon.