ArduPilot/ardupilot #5024: Jump Loiter-AltHold NOT solved + some other problems/errors on EKF that lead to land the copter

andrea25, 2016-10-19:
#### Issue details
Hi,
I have done some tests with AC3.4-rc6.
I think I have found two different bugs.

1-When we change from AltHold to Loiter the copter still fly high. To high.
It do not happen everytime, only the first time.
This is the video
https://www.facebook.com/projectEMS/videos/880820752018402/
Try to wath from 1:40. It is the first mode change (from althold to loiter)
This is the log:
https://drive.google.com/file/d/0ByiQ-PFiGc1MRXN0cHV3YnlaSlk/view?usp=sharing

2-Some problems on EKF lead me to land the hexacopter.
I don't why, but the Mission Planner had send me some warning about EKF.
And then the copter landed itself.
I can't change the landing process after it was engaged for failsafe.
This is the log:
https://drive.google.com/file/d/0ByiQ-PFiGc1MVGJjQkhPV0VVNFk/view?usp=sharing

#### Version
AC3.4-rc6

#### Platform
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover

#### Airframe type
Hexa X6

#### Hardware type
PixFalcon from HobbyKing

#### Logs
See the description

govsux, 2016-10-19:
re- I can't change the landing process after it was engaged for failsafe. 
Do you mean even if you try to manually change to another mode to stop the landing it doesn't respond??

rmackay9, 2016-10-19:
Thanks very much for testing AC3.4-rc6.  The sudden jump during mode change definitely looks like a bug so I will investigate this.

Normally we ask that support issues be added to the discourse server.  http://discuss.ardupilot.org/c/arducopter/copter34.  Github issues list is just for confirmed bugs and enhancement requests.  Sorry to be strict but if we don't do this then we end up with the issues list turning into a support forum.

thanks again.

closed 2016-10-19 by rmackay9