ArduPilot/ardupilot #5028: Led status problem with fence activation/desactivation

thecodingfrog, 2016-10-20:
### Issue details
If you disable fence while no GPS fix, led turn to blue and system is armable,then if you enable fence and still don't have GPS fix the led should turn back to yellow, but it remains blue

#### Version
3.3.3

#### Platform
[ * ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
Quad

#### Hardware type
Pixhawk

OXINARF, 2016-10-22:
Yellow means pre-arm checks have failed. Blue means GPS isn't fixed. Green means GPS is ready to go.

You are correct that led behavior isn't consistent: GPS can make pre-arm checks fail (and hence led is yellow) and if for one time the checks pass then it won't be yellow again and only blue and green are shown.

WickedShell, 2016-10-22:
@OXINARF not sure I'm following something here. On plane I can fail/not fail arming checks as  many times as I want and go back and forth between blinking  green led to blinking yellow LED. Is this something copter/arming specific or a problem with the GPS side?

OXINARF, 2016-10-22:
It's in Copter and I haven't checked other vehicles. The issue is that once it passes pre-arm checks it doesn't go back. GPS pre-arm checks are still run in that case but it only switches between green and blue.

WickedShell, 2016-10-22:
Thats definitely not the case for plane. (I've used the LED thing quite a lot when testing/purposefully failing the GPS config checks).

OXINARF, 2016-10-22:
Yes, just confirmed and both Plane and Rover use AP_Arming which always runs pre-arm checks, whether they have passed or not. Copter doesn't run them once they pass.

thecodingfrog, 2016-10-22:
By the way, many info are passed by GCS_Mavlink but not prearm check. On my v3.3.3 custom firmware I've added it to heartbeat message so my led controler is more consistent

rmackay9, 2016-11-19:
This is a duplicate of this other issue:https://github.com/ArduPilot/ardupilot/issues/4918.

This one has more discussion so I'll close the other issue.

IamPete1, 2021-08-15:
This is fixed in master.

closed 2021-08-15 by IamPete1