ArduPilot/ardupilot #5033: Compilation fails if fence is disabled

night-ghost, 2016-10-21:
#### Issue details
If fence disabled by

#define AC_FENCE               DISABLED

then compilation fails:

Copter.cpp: In constructor 'Copter::Copter()':
Copter.cpp:79:31: error: 'fence' was not declared in this scope
     avoid(ahrs, inertial_nav, fence, g2.proximity),
                               ^
make: *** [/tmp/ArduCopter.build/Copter.o] Ошибка 1

#### Version
3.4rc6

#### Platform
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover

rmackay9, 2016-10-22:
Yes, similarly I think Proximity can't be disabled without producing a compile error.  I think we need to make some changes to pass some objects (like proximity) into other objects through a new function instead of using the constructor.  This would allow us to not provide proximity if it's not present.

rmackay9, 2016-10-22:
I've cleared the AC3.4 milestone 'cuz I don't think it's a requirement that we get this fixed before the AC3.4 release.

OXINARF, 2016-10-22:
@rmackay9 It isn't. I put that milestone because the Avoidance library looked easy to fix.

shortstheory, 2018-03-14:
Compilation no longer fails with AP_FENCE disabled in the Copter module.

Hence, the preprocessor statement in config.h:671 can be removed:

```
#if AC_AVOID_ENABLED && !AC_FENCE
  #error AC_Avoidance relies on AC_FENCE which is disabled
#endif
```

shortstheory, 2018-03-15:
@rmackay9 IMO it would be easier to just change:

```
#if AC_FENCE == ENABLED
    AC_Fence fence{ahrs};
#endif
```
to

```
    AC_Fence fence{ahrs};
    fence.enable(false);
#if AC_FENCE == ENABLED
    fence.enable(true);
#endif
```

Otherwise, I feel that a lot of changes would have to be made in overloading constructors.

rmackay9, 2019-05-18:
There are nice messages now which guide the developer as to which features need to be disabled is fence is disabled.  So it seems these are avoidance and follow mode.

I think handling the message well resolve this issue so I think I'll close it.

closed 2019-05-18 by rmackay9