ArduPilot/ardupilot #5049: Problem with "AP_Compass_BMM150"

guantau, 2016-10-23:
AP_Compass_BMM150 is complied under Raspbian linux and applied on the Raspberry Pi 2. Basically, BMM150 is connected through I2C bus. The device address `CHIP_ID_REG 0x12/0x13` and `CHIP_ID_VAL 0x32` can be read correctly. However,  `_load_trim_values()` (on line 164) return all zeros and `_update()`(on line 253)  of DATA_X_LSB_REG return all ones. 

The Raspberry Pi 2 used for test works well with navio2. So, it is probably not a hardware problem. Can it be fixed?
 
@lucasdemarchi @zehortigoza @guludo 
Thanks a lot.

lucasdemarchi, 2016-10-24:
It may be a problem with the board in which BMM150 is.  I took a quick look in the code and couldn't find anything suspicious.  This is tested and flying with Intel Aero Compute board and the part is really BMM150, not BMC150. Are you sure you got the right part?  What board are you using for testing?

closed 2016-11-07 by guantau

lucasdemarchi, 2016-11-07:
@guantau any reply why you are closing this? Was this a bug or what?

guantau, 2016-11-07:
@lucasdemarchi I examine my hardware and find that there's a problem with the BMM150 I use. So it's not a bug. Thank you very much for your help!

R-Lefebvre, 2016-11-13:
guantau, would you mind giving a bit of detail on your hardware problem?  Just curious as I am considering this chip for a system.  Just wanted to know if the hardware problem is a general problem with the chip, of a specific problem restricted to your particular hardware.