ArduPilot/ardupilot #5052: Erroneous wind estimation when coming back from GNSS restricted area

suraqis, 2016-10-23:
#### Issue details
When fixed wing UAV come out from GNSS restricted area and GNSS signals are not stable and precise errors in wind speed and direction estimation can occur. As far as I understand the process when GNSS signals recover after loss in GNSS restricted area they can be unstable and jittering for a while and ground speed computation can be erroneous and thus wind estimation too. Some hold delay should be applied to ground speed and wind estimator for GNSS signals to became consistent. 

#### Version
ArduPlane 3.6.0

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[x] Plane
[  ] Rover

#### Airframe type
Fixed wing UAV with airspeed sensor

#### Hardware type
Pixhawk

#### Logs
Unfortunately can't provide logs. It failed to write to card (((

Naterater, 2019-01-04:
I'm not sure I see the value in a more accurate wind speed estimate if we are relying on airspeed anyway? Does this cause some erratic behavior?

priseborough, 2019-01-06:
Please provide a log illustrating the behaviour.

EShamaev, 2019-01-07:
@suraqis You always have at least the GS logs, these are written to PC

IamPete1, 2021-08-15:
With no log I don't think there is any value in keeping this issue open.

closed 2021-08-15 by IamPete1