ArduPilot/ardupilot #5062: Add parameter for disabling Stick based Compass calibration

wjax, 2016-10-25:
#### Issue details
Stick initiated compass calibration is going to be a new feature of Copter. There should be a parameter associated to allow users and manufacturers to disable it.
If it is enabled with no disabling option there will be a lot of people inadvertently activating it and increasing the APM complexity perception.

#### Version

#### Platform
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

#### Airframe type

#### Hardware type

#### Logs

WickedShell, 2016-10-25:
@wjax Worth noting that only copter has the gesture interface. I believe solo has the same one, but it is explicitly not supported on plane/rover as the gestures required are highly likely to have safety problems causing power to be applied to the motor.

OXINARF, 2016-10-26:
@wjax I'm not sure I'm following your concern. You need to have the sticks at max throttle and max yaw for two full seconds to activate the compass calibration. Also you need to be in the disarmed state.
What is the issue you are seeing here?

wjax, 2016-10-26:
Francisco,

The concern is about complexity.
Those UAV manufacturers out there keep away from APM because this kind of things. We are one of those manufacturers and try and want to stay with APM but it is difficult.
I have always expressed my concern about the high level of complexity that APM is getting. Not the internal one which is fine, but the one that faces the user in a daily use.

APM is like the Linux of the UAV autopilots. Many options and possibilities but difficult for a user (client) that wants zero-problems and just get the work done reliably.

Manufacturers don't want their clients to be able to change out-off-the-factory settings like the so critic and difficult to get done compass calibration. You instruct them to not touch certain places in Mission planner but stick calibration is imho a potential source of problems.

I do not see the need to do stick based field calibration at all, by the way. It can only be required if you have moved from continent or you have changed your internal setup way much. Things that for most people does not occur often. And if it happens could be launched with gcs.
If people and devs think that stick based calibration is OK, then OK. 
But my request is to have the possibility to disable it's activation via stick movement.

That's my concern

OXINARF, 2016-10-27:
I understand your point. But when you have a software that wants to have many features you are bound to find these problems. On the other hand I fully believe in choice and there's nothing that annoys me more than when software can't work the way I want it to.

You say that compass calibration is difficult to get done, but if that is true then it is ArduPilot's fault because it shouldn't be hard.

Regarding the need for stick calibration: people may want to calibrate without having a GCS around. It isn't essential (it wasn't here until now) but it a nice thing to have.

Regarding manufacturers: we want to keep everything in master as far as possible, but I think there's nothing in manufacturers doing their own fork with a couple of patches on top - that will always be easy to keep updated from master.

Anyway, this is a valid request and I just wanted to understand your concern.

magicrub, 2016-10-27:
Fyi, plane has stick arming and disarming but not compass stuff.

wjax, 2016-10-28:
Thanks Francisco

Sorry to be in the side of business and client thinking :)
I think that sometimes Ardupilot needs to think a little more in the KISS strategy. If companies have to use Ardupilot, some attention should be paid to simple interaction, intuitive reactions and procedures and a large etc.

I can provide such feedback if you want.

Regards!

peterbarker, 2016-10-28:
On Fri, 28 Oct 2016, Jesus wrote:

> Sorry to be in the side of business and client thinking :)

No, thanks, we appreciate receiving your viewpoint!

Ironically, extra parameters are usually seen in ArduPilot as *increased* 
complexity, not *decreased* complexity!

Peter

wjax, 2016-10-28:
They are in fact as you say. I do not have a problem with more parameters as they will be "touched" by the manufacturer. Not by the client.
So if those parameters allow that Ardupilot fit in any need, they are welcomed!

OXINARF, 2016-10-28:
Like Peter said, the input is definitely appreciated. Let me just say that one of the reasons we set up the Partners program is precisely for this, hearing what the commercial entities connected to ArduPilot have to say. Obviously we can't make ArduPilot better for companies if there are no hired resources to do it, because otherwise all contributions are volunteer - and it isn't fair to reject a PR just because it is missing a parameter to disable the functionality it introduces.

rmackay9, 2016-10-29:
I agree wjax that we need to keep KISS in mind... we definitely do at times.. there are at least a couple of changes in the past few months that we haven't put in because they appear to be quite complex and would only be used by a tiny tiny fraction of users.

I've also agree that the compass calibration should not, in general, be needed very often.  Even moving continents shouldn't mean a compass calibration is required because the "declination" is looked up from a small on-board database, the "inclination" is captured at start-up and again early in flight.

Anyway, it's in now, let's see how many people actually accidentally trigger it before we put a patch in to protect against this.  We have this sort-of-golden-rule (at least in copter) that we don't fix a problem unless we're sure we actually have the problem.

R-Lefebvre, 2016-11-01:
I have to agree with Jesus here.

I will encourage my customers to never operate the system without a GCS connected, as the system is simply too complicated.  And stick gestures just pile on top of that.

Jesus, you'll probably have to prepare a custom firmware.

hdtechk, 2018-02-06:
I have to say after more than a year using stick gesture for onboard calibration.    I love it.   Had lots of issues onboard  calibration started from ground control station.     But when I leave  tablet /laptop in car walk out in field.   And start calibration with full throttle and full right yaw.    I have 100%  success rate.

rmackay9, 2018-02-06:
I'm not even sure we've documented anywhere on the wiki how to kick off the compass calibration from the sticks.  I haven't heard any reports of users accidentally triggering it.

Pedals2Paddles, 2018-02-06:
In use it all the time. I haven't initiated a compass Cal with the GCS in a long time actually. But I can definitely understand wanting to do some idiot proofing when a customer is involved.

Maybe copter should have a new flight mode called service mode? Would allow for this and potentially many more gestures.

hdtechk, 2018-02-06:
It's not documented anywhere.   I will post link to Jonathans video  in ArduPilot  wiki gitter if you would like I thought about putting in practice for it but don't know how to make it look good.
I like your idea of service mode  Matt. We use service mode on harleys canbus systems to keep security alarms from going off during work

hdtechk, 2018-02-06:
PR not practice stupid spell check on phones

PR #18338 (2021-08-23): Copter/Blimp: remove stick gesture compass calibration start and stop
Fixes #5062

Flight options is not a great place for this, but I don't think its worth adding a new options param just for this.
closed 2021-08-23 by rmackay9