ArduPilot/ardupilot #5077: problem in make ardupilot in windows

hamididari, 2016-10-27:
hi guys

i try to make ardupilot in windows from [this](http://ardupilot.org/dev/docs/building-px4-with-make.html)
but i got this err 
[image]

any idea about my problem ?
i use pixhawk and ArduCopter
thanks

khancyr, 2016-10-27:
hi did you install lastest toolchain and  use git submodule update --init --recursive ? 
The error looks like missing mavlink header, but it should have been generated before compiling

hamididari, 2016-10-27:
i uninstall my px4 tool chain and install the last version and make again but i got the same err again

khancyr, 2016-10-27:
Sorry I don't use windows for that, a more qualified person should answer.
If you change the toolchain to lastest, do a make clean to remove previous compiled filed and retry

rmackay9, 2016-10-27:
Maybe try a "make px4-clean" and "make clean" and/or delete the /modules/PX4Firmware and mavlink directories.  I compile all the time with Windows using the latest toolchain which is here:http://firmware.ap.ardupilot.org/Tools/PX4-tools/

OXINARF, 2016-10-27:
This is more of a support question and not a bug so I'm closing this. Feel free to post in our forum (http://discuss.ardupilot.org) or chat with us in Gitter (http://gitter.im/ArduPilot/ardupilot) if you still need help.

closed 2016-10-27 by OXINARF