ArduPilot/ardupilot #5080: ROVER: can't clear mission until it is finished

khancyr, 2016-10-27:
#### Issue details
Hi,

On Rover, we can't clear mission until the mission is really finished. Changing for auto to other mode isn't consider at mission stop.

How to see the bug : 

sim_vehicle.py -v APMrover2 -m --mav10 --console --map -w
arm throttle
wp load Tools/autotest/ap1.txt (or whatever mission)
mode auto

(wait mission to start)

mode hold
(wait it stop)
wp clear
 ----> rejected !
wp list
(waypoint mission is still here)

mode guided
(wait it stop)
wp clear
 ----> rejected !
wp list
(waypoint mission is still here)

mode manual
(wait it stop)
wp clear
 ----> rejected !
wp list
(waypoint mission is still here)

mode auto

(wait mission to finish)
wp clear
 ----> accepted !
wp list
(waypoint mission is empty)

#### Version
From rove 2.50 to  https://github.com/ArduPilot/ardupilot/commit/e8b9c815fcf9c1bcd8fa87f6aa8637bd20d64b72

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[ X] Rover

#### Hardware type
Present in both pixhawk and sitl

#### Logs
_Please provide a link to any relevant logs that show the issue_

khancyr, 2016-10-27:
In https://github.com/ArduPilot/ardupilot/blob/master/APMrover2/system.cpp#L261
is missing something like https://github.com/ArduPilot/ardupilot/blob/152edf71891146abc588b86e5f4149405b794da1/ArduCopter/flight_mode.cpp#L263

magicrub, 2016-10-27:
> In https://github.com/ArduPilot/ardupilot/blob/master/
> APMrover2/system.cpp#L261
> is missing something like https://github.com/ArduPilot/ardupilot/blob/
> 152edf71891146abc588b86e5f4149405b794da1/ArduCopter/flight_mode.cpp#L263

I think plane is the same way. You can't clear a mission if currently
executing a mission. However, you can overwrite a mission so I guess you
should be able to clear it, although what do you expect it to do? You're in
auto and you've cleared the mission. Now what?

khancyr, 2017-04-14:
Solve by ROVER: stop mission when leaving AUTO #5739

closed 2017-04-14 by khancyr