ArduPilot/ardupilot #5112: Copter-3.4: PPM sum not working on Pixracer?

rmackay9, 2016-10-31:
Some Pixracer users are reporting that AC3.4 is not accepting PPM input.

rustutzman, 2016-11-01:
I bought my Pixracer back in January. The only version info I see is Pixracer v1. Glad to help with anything I can.

rmackay9, 2016-11-01:
Here's a discussion in the support forums: http://discuss.ardupilot.org/t/ppm-with-pixracer/12448/6

OXINARF, 2016-11-01:
This has been fixed in master and backported to the Copter-3.4 branch. Copter 3.4.1 is on its way out.

closed 2016-11-01 by OXINARF

rmackay9, 2016-11-01:
This is fixed in master and is going out with AC3.4.1.

brizey02, 2017-01-16:
Seems still broken on ez-uhf, pixracer and taranis

gregd72002, 2017-01-18:
@brizey02 I think that some pixracer fundamentally do not support inverted ppm signal. I have had similar issues with mine and the only way was to use a simple inverter in order to normalize it. You might want to give it a try

https://cloud.githubusercontent.com/assets/16255054/16666545/6bce6128-4488-11e6-9844-7b29ce07776d.png