ArduPilot/ardupilot #5120: Feature Request: ESC (BLHeli) USB Passthrough

Frontier314, 2016-11-01:
#### Issue details
Feature Request: ESC (BLHeli) USB Passthrough

#### Version
Any

#### Platform
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover

#### Airframe type
Quadcopter, Hexacopter

#### Hardware type
PixRacer

#### Logs

magicrub, 2016-11-01:
can you give a few more details as to what USB Passthrough means?

juvinski, 2016-11-01:
Hi @magicrub,

I believe the Passthrough means you can change parameters or even flashing BLHeli ifrmware using PixRacer as an programming interface/adapter.

magicrub, 2016-11-01:
does this have anything to do with USB?

Frontier314, 2016-11-01:
Apologies for not explaining, I thought it was a known feature of other firmware.

USB passthrough allows any FC running CleanFlight (or Betaflight or iNav) to use the BLHeli suite to flash and program ESCs while they are connected to the FC. This is extremely handy for miniquads/racers where the ESC servos most of the time are connected (or even soldered directly) to the ESC.

This "passthrough" feature is provided by the CleanFlight/BetaFlight/iNav firmware and it is portable among supported boards (i.e you can do it on SP Racing F3, Naze32 or even CC3D boards). Think of it like the GPS-passthrough feature APM/PixHawk/PixRacer boards have, where you can flash and program your uBlox GPS while it is connected to the FC, without having to use an FTDI adapter/cable.

Currently I am running AC3.4.1 on two PixRacers I have, one mounted on an S500 frame and the other on a "Robocat"-like 270mm racer frame (a cheap one from eBay). It is totally PITA to change ESC parameters on the racer without dismantling everything, since everything is so crammed in order to fit.

Again, thank you very much for the interest.

mkschreder, 2016-12-22:
It is basically forwarding of traffic between the main serial port and the esc ports that bitbang the blheli protocol over the same wire. It's pretty useful for changing esc parameters but it does require some control over the pwm pins to be able to use them for the 1 wire serial communication that the escs use.

lvale, 2016-12-23:
@magicrub Tom, this topic has been discussed at dev calls with Tridge, and is somewhat "tied" with the DMA to pin issue also relevant for the support of DShot

magicrub, 2017-01-10:
@lvale do you have any issues/PRs you can cross link to here?

lvale, 2017-01-10:
probably this https://github.com/ArduPilot/ardupilot/pull/5519

tridge, 2017-01-10:
I'd very much like to support this, although I can't promise when I will have time to work on it

Frontier314, 2017-01-11:
Thank you all very much for the contribution - especially @tridge - I am really hoping that you'll manage to find some time and work on it as it would be a great feature for ArduPilot.

dgrat, 2017-05-31:
@tridge @magicrub @Frontier314 
I don't know whether it is possible for some boards to read from the RCOUT pins. 
The ESC configuration and update goes in Cleanflight over a one wire connection allowing the ESCs to send some data. 
Not sure whether e.g. PixRacer can read from the RCOUT pins. If so, I would like to implement something like this.

BVN99, 2017-07-05:
@Frontier314 does BLHeli suite have support for usb pass through for PX4? If not how do I  connect my FC {pixhawk mini} to the suite so I can flash and configure my blHeli escs?

Frontier314, 2017-07-06:
No, it does not support USB passthrough for PX4, only for CleanFlight/Betaflight/iNav targets.

dgrat, 2017-07-07:
Can someone comment to question whether one can read from the RCouts?
Alternatively one could connect the ESCs maybe to the UARTs?!

mkschreder, 2017-07-07:
Somebody could check the pixhawk schematic perhaps. I don't really why not
because most pins on stm32 work well as inputs too. But it could be tricky
to add from software arch perspective..

dgrat, 2017-07-07:
Currently I am not even sure what the ESCs send back. Guess it is PWM: https://github.com/iNavFlight/inav/blob/29e67bb1238e4e83c28a5e1b454df4b5f1ed3d4a/src/main/io/serial_4way.c
One  just would need an RCIn for the RCOuts and write a tool for passthrough to allow configuration and flash.

Engineereric, 2017-09-15:
its normal serial data, if you look at the BLHeli_s code its on a normal serial port, TX and RX are essentially tied together and its half-duplex.

dgrat, 2017-09-15:
@Engineereric @tridge 
I guess one would need to implement the DShot protocol. This will probably require DMA. Does the Pixracer/Pixhawk have sufficient resources for that? I am not the e-tech specialist. However, it would be interesting to know.

Engineereric, 2017-09-15:
No DShot protocol defines the 1's and 0's by length of pulses unlike serial that is high or low. You can "Bit Bash" the serial for programing BLHeli_s as the pass-though in Betaflight even works on an old F1 CC3D board.

tridge, 2018-03-22:
I have ESC pass-thru config working on ArduPilot now, and have used it successfully with both BLHeli_32 and BLHeli_S ESCs.
Please see my comments in issue #5300 for more details.

rmackay9, 2018-07-03:
I think this issue is complete in master at least for Copter.  maybe we can close this issue?

closed 2018-07-03 by Frontier314