ArduPilot/ardupilot #5128: [PLANE][VTOL] Compass motor calibration for VTOL Planes

andrea25, 2016-11-03:
#### Issue details
I think that having the possibilty to do Compass Motor calibration on Plane can be very useful.

I have issues only during high throttle input while using quad motors. Mission planner say "compass variance" and I see the "toilet bowling" so I think that maybe compass motor can help. I had searched but I didn't find the compass motor option for plane.

Maybe it is something that have to be coded on the Mission Planner?

PS: if this is not the right place to post this request, please move it where you think that is more appropiate. I am not too familiar with GitHub.

#### Version
ArduPlane 3.7.1

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ X ] Plane
[  ] Rover

#### Airframe type
VTOL QuadPlane

#### Hardware type
Pixfalcon

#### Logs
No logs

marcmerlin, 2017-06-10:
Also requested here http://discuss.ardupilot.org/t/compass-motor-calibration-for-quadplane/18265

and I'll add my wish that it were supported in regular planes too. They totally can impact the compass with the motor in a small build.

DavidIngraham, 2018-01-09:
+1
A note: because compass-mot corrections happen in the compass library, it is possible to flash copter to perform the calibration, flash plane, copy the params, and re-enable compass-mot corrections. However it would be much more convenient to either:

- support onboard compass-mot cal in Plane
or 
- Provide a script that can calculate the offsets from a log.

IamPete1, 2025-03-08:
We don't support throttle based compensation on Plane but current compensation works. We now have a WebTool to calculate the parameters from a log. https://firmware.ardupilot.org/Tools/WebTools/

closed 2025-03-08 by IamPete1